siemens/ros-sharp #229: Bloced in tutorial 2.1 under melodic(under ubuntu 18.04)

zenoCaiCai, 2019-07-15:
## I have a question: ##

I am new to ROS system and Unity engine. 

I am trying to using a slam system to offer localization for a VR project. I want to use ros# to transport information from slam into Unity.

I know little about URDF and turtlebot.

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I'm blocked at step 2.1 of the wiki of this project. 

I need to run the code below: 
`$ roslaunch file_server publish_description_turtlebot2.launch`

However, I am using Ubuntu 18.04 and the corresponding ROS version: melodic. I can only install turtlebot3 instead of turtelbot2 under this version of melodic in the tutorial step 1.5.

So I cannot launch the launch file the above-mentioned, and got error messages as below:
```
[roslaunch][INFO] 2019-07-11 16:23:37,359: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2019-07-11 16:23:37,361: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2019-07-11 16:23:37,361: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'file_server', 'publish_description_turtlebot2.launch']
[roslaunch][INFO] 2019-07-11 16:23:37,361: roslaunch env is {'XDG_SESSION_TYPE': 'x11', 'LC_PAPER': 'zh_CN.UTF-8', 'SHELL': '/bin/bash', 'XDG_DATA_DIRS': '/usr/share/budgie-desktop:/usr/local/share:/usr/share:/var/lib/snapd/desktop', 'MANDATORY_PATH': '/usr/share/gconf/budgie-desktop.mandatory.path', 'CLUTTER_IM_MODULE': 'xim', 'NVM_BIN': '/home/vrlab01/.nvm/versions/node/v10.16.0/bin', 'TEXTDOMAIN': 'im-config', 'XMODIFIERS': '@im=ibus', 'INVOCATION_ID': '9c7cd3935d2d48688a12b2398f07af79', 'XDG_RUNTIME_DIR': '/run/user/1000', 'LC_ADDRESS': 'zh_CN.UTF-8', 'PYTHONPATH': '/home/vrlab01/Desktop/rostest/catkin_ws/devel/lib/python2.7/dist-packages:/home/vrlab01/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'DBUS_STARTER_BUS_TYPE': 'session', 'XDG_SESSION_ID': 'c1', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus,guid=33ec64850c3c3de98793dc505d26e0be', 'DESKTOP_SESSION': 'budgie-desktop', 'GTK_MODULES': 'gail:atk-bridge', 'TURTLEBOT_3D_SENSOR': 'kinect', 'LC_NAME': 'zh_CN.UTF-8', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'LC_NUMERIC': 'zh_CN.UTF-8', 'DBUS_STARTER_ADDRESS': 'unix:path=/run/user/1000/bus,guid=33ec64850c3c3de98793dc505d26e0be', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', '_CE_M': '', 'XDG_CURRENT_DESKTOP': 'Budgie:GNOME', 'TILIX_ID': '6dce8bc0-279d-4180-8bca-61e953548507', 'USER': 'vrlab01', 'XDG_VTNR': '7', 'CMAKE_PREFIX_PATH': '/home/vrlab01/Desktop/rostest/catkin_ws/devel:/home/vrlab01/catkin_ws/devel:/opt/ros/melodic', 'MANAGERPID': '1232', 'XAUTHORITY': '/home/vrlab01/.Xauthority', 'JOURNAL_STREAM': '9:20045', 'SESSION_MANAGER': 'local/vrlab01:@/tmp/.ICE-unix/1244,unix/vrlab01:/tmp/.ICE-unix/1244', 'LC_MEASUREMENT': 'zh_CN.UTF-8', 'QT_QPA_PLATFORMTHEME': 'gtk2', 'LD_LIBRARY_PATH': '/home/vrlab01/Desktop/rostest/catkin_ws/devel/lib:/home/vrlab01/catkin_ws/devel/lib:/opt/ros/melodic/lib', 'QT_ACCESSIBILITY': '1', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'XDG_SESSION_DESKTOP': 'budgie-desktop', 'GDMSESSION': 'budgie-desktop', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'ROS_PACKAGE_PATH': '/home/vrlab01/Desktop/rostest/catkin_ws/src:/home/vrlab01/catkin_ws/src:/opt/ros/melodic/share', 'TURTLEBOT_BASE': 'roomba', '_': '/opt/ros/melodic/bin/roslaunch', 'LC_IDENTIFICATION': 'zh_CN.UTF-8', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-budgie-desktop:/etc/xdg', 'ROS_MASTER_URI': 'http://localhost:11311', 'COLORTERM': 'truecolor', 'LC_TIME': 'zh_CN.UTF-8', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/vrlab01', 'QT4_IM_MODULE': 'xim', 'PKG_CONFIG_PATH': '/home/vrlab01/Desktop/rostest/catkin_ws/devel/lib/pkgconfig:/home/vrlab01/catkin_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig', 'NVM_DIR': '/home/vrlab01/.nvm', 'HOME': '/home/vrlab01', 'DISPLAY': ':0', 'LANG': 'en_US.UTF-8', 'CONDA_SHLVL': '1', 'ROS_LOG_FILENAME': '/home/vrlab01/.ros/log/4836a6a0-a3ab-11e9-9856-30b49eba19df/roslaunch-vrlab01-13978.log', 'LC_MONETARY': 'zh_CN.UTF-8', 'CONDA_PREFIX': '/home/vrlab01/anaconda3', 'CONDA_DEFAULT_ENV': 'base', 'ROS_PYTHON_VERSION': '2', 'GTK_IM_MODULE': 'ibus', 'VTE_VERSION': '5202', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'DEFAULTS_PATH': '/usr/share/gconf/budgie-desktop.default.path', 'PAPERSIZE': 'a4', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'ROS_DISTRO': 'melodic', 'NVM_CD_FLAGS': '', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/vrlab01/Desktop/rostest/catkin_ws/devel/share/common-lisp:/home/vrlab01/catkin_ws/devel/share/common-lisp', 'QT_IM_MODULE': 'ibus', 'LOGNAME': 'vrlab01', 'XDG_SEAT': 'seat0', 'PATH': '/opt/ros/melodic/bin:/home/vrlab01/.nvm/versions/node/v10.16.0/bin:/home/vrlab01/anaconda3/bin:/home/vrlab01/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/vrlab01/.dotnet/tools', 'SSH_AGENT_PID': '1376', 'TERM': 'xterm-256color', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'QT_STYLE_OVERRIDE': '', 'TURTLEBOT_STACKS': 'circles', 'IM_CONFIG_PHASE': '2', '_CE_CONDA': '', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'TEXTDOMAINDIR': '/usr/share/locale/', 'CONDA_PROMPT_MODIFIER': '(base) ', 'LANGUAGE': 'en_US:en', 'OLDPWD': '/home/vrlab01', 'GDM_LANG': 'en_US', 'LC_TELEPHONE': 'zh_CN.UTF-8', 'SHLVL': '1', 'PWD': '/home/vrlab01/Desktop/rostest/catkin_ws', 'ROS_VERSION': '1', 'CONDA_EXE': '/home/vrlab01/anaconda3/bin/conda'}
[roslaunch][INFO] 2019-07-11 16:23:37,361: starting in server mode
[roslaunch.parent][INFO] 2019-07-11 16:23:37,362: starting roslaunch parent run
[roslaunch][INFO] 2019-07-11 16:23:37,362: loading roscore config file /opt/ros/melodic/etc/ros/roscore.xml
[roslaunch][INFO] 2019-07-11 16:23:37,451: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2019-07-11 16:23:37,451: loading config file /home/vrlab01/Desktop/rostest/catkin_ws/src/file_server/launch/publish_description_turtlebot2.launch
[roslaunch][INFO] 2019-07-11 16:23:37,482: Added node of type [rosbridge_server/rosbridge_websocket] in namespace [/]
[roslaunch][INFO] 2019-07-11 16:23:37,482: Added node of type [rosapi/rosapi_node] in namespace [/]
[roslaunch][INFO] 2019-07-11 16:23:37,482: Added node of type [file_server/file_server] in namespace [/]
[roslaunch][ERROR] 2019-07-11 16:23:37,483: RLException: Invalid <arg> tag: turtlebot_description
[... cut ...]
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/vrlab01/Desktop/rostest/catkin_ws/src
ROS path [2]=/home/vrlab01/catkin_ws/src
ROS path [3]=/opt/ros/melodic/share. 

Arg xml is <arg default="$(find xacro)/xacro.py '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'" name="urdf_file"/>

[rospy.core][INFO] 2019-07-11 16:23:37,483: signal_shutdown [atexit]
```

Hope someone could help me or offer some hint. Thanks for you guys!

** My environment: **
ROS: melodic
Ubuntu: 18.04

zenoCaiCai, 2019-07-15:
By the way, I tried the ROS# tutorial under 16.04 with turtlebot2, and all things went fine. But I am still wondering if I could use this under 18.04 because the distro I am using will not offer stable release of 16.04 since now.

zenoCaiCai, 2019-07-16:
# Update: I tried turtlebot3_description file.

I have knew that turtlebot_description package has not release to melodic yet, and it may not have that  forever..... Then I found the corresponding version to turtlebot3, it's turtlebot3_description package, which support melodic version.

I changed some code in file `ROS/file_server/launch/publish_description_turtlebot2.launch`, the 25th line from 
```
<arg name="urdf_file" default="$(find xacro)/xacro.py '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'" />
``` 
into 
```
<arg name="urdf_file" default="$(find xacro)/xacro.py '$(find turtlebot3_description)/urdf/turtlebot3_burger.urdf.xacro'" />
```
Which means, the luanch file will use `turtlebot3_burger.urdf.xacro` file in package turtlebot3_description instead of the origin turtlebot2 urdf file. After pip install some dependencies, I really did it, I ran the launch file successfully, and got bash info below:
```
[INFO] [1563263144.433879]: Rosbridge WebSocket server started on port 9090
```

However, after I opened RosBridgeClient and then changed the corresponding address, I clicked the button `Read Robot Description`, then I got the `Connected` text into green, while others still red.

I think this may occurred because of the corresponding code somewhere else are still dealing with urdf file as turtlebot2 version, but I cannot dig more. If there is someone who can help me, please contact me, thanks!

zenoCaiCai, 2019-07-16:
# Tap screenshot image

[image]
I run both ROS melodic and Unity3D under the same Ubuntu 18.04 system.
Left side is Unity window, right side is ROS bash with the launch file output. Once I click the `Read Robot Description` button, right side output will add one client.

berkayalpcakal, 2019-09-17:
Hello @zenoCaiCai 

Once I was able to build turtlebot2 related packages with ROS Melodic on Ubuntu 18.04, but you have to build everything from source files by yourself. 

So what is the situation with your issue now? I guess you are already aware of the reasons and the solutions to the issue regarding incompatibility of turtlebot2 packages to ROS Melodic. 

I think you can open another issue with a suitable issue title.

closed 2019-09-17 by berkayalpcakal

HieskarSG, 2023-08-14:
Hello! 
Thank you so much for talking about this topic. I'm stuck in the same place.

I don't know if you could solve this. But if so, how did you do it?