siemens/ros-sharp #232: Unity errors when adding libraries in the Asset folder

Scarf195, 2019-07-21:
- [ x ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.

* [ x ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ x ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
 Hello! Thank you so much for sharing the source code and writing the detailed tutorials, they were of excellent help as well as very interesting. I'm new using ROS# and I'm trying to make an Unity template putting all sources of the RosBridgeClient and Urdf libraries in the Unity Assets folder and deleting RosbridgeClient.dll and Urdf.dll from the Plugins folder instead, this to better understand how the library works (following also what said in the issue #169).

Once I open Unity gives me many warnings and errors and I can't find a solution for that.
Could you explain me what I have to add specifically in the asset?

Thanks.

berkayalpcakal, 2019-09-17:
Hello @Scarf195 

Any progress so far? Maybe describing the errors you get in detail might help for better understanding?

PymZoR, 2019-09-27:
My setup :
- Ubuntu 18.04
- Unity 2019.3.08a
- .NET 4.x

Here is what I get when copying the source of RosBridgeClient inside my project and removing the .dll 

[image]

Seems that part of the problem is namespace clashing between internal classes like `Publisher` with ROS# `Publisher` amongst others