siemens/ros-sharp #235: Publish and Subscribe Node Between ROS and Unity IOS Application

cerenoozturk, 2019-07-23:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

I imported rossharp to my unity project. By rosbridge I am connecting Unity to ROS . I am successfully connecting and publishing node from Unity to ROS. Then I tried to make the same thing from my Iphone by building an Unity application for Iphone. When I run Unity app I am connecting to ROS but I cant publish any node from application to ROS. 
I think the problem is coming from there:

**NullReferenceException: Object reference not set to an instance of an object.
  at RosSharp.RosBridgeClient.PoseStampedPublisher.UpdateMessage () [0x00000] in <00000000000000000000000000000000>:0** 

How can I fix this problem? Is there any other way to publish and subscribe node between ROS and Unity IOS application?

---

TomSevestre, 2019-07-24:
I have the same problem on my side but with an Android application. My application is connecting to ROS but can't subscribe or publish to ROS topics.

Scarf195, 2019-07-24:
Hi, I had the same problem with my Andoid application! I think it's about the Json file. Follow the issue #188 and try.

TomSevestre, 2019-07-25:
> Hi, I had the same problem with my Andoid application! I think it's about the Json file. Follow the issue #188 and try.

I followed your issue, but there is one thing that I don't understand you talk about a file Communcation.cs but I can't find it anywhere, so where is this file ?

cerenoozturk, 2019-07-25:
The Communication.cs file is in ros-sharp/Libraries/RosBridgeClient . However it is updated 27 days ago, I am not sure if we still need to change the code in Communication.cs.

TomSevestre, 2019-07-25:
> The Communication.cs file is in ros-sharp/Libraries/RosBridgeClient . However it is updated 27 days ago, I am not sure if we still need to change the code in Communication.cs.

But I never download these Libraries, I just downloaded the Asset on the Unity Asset Store. So if I download it where do I need to put these Libraries ? I didn't in the wiki where we need to put this Libraries

Scarf195, 2019-07-25:
> The Communication.cs file is in ros-sharp/Libraries/RosBridgeClient . However it is updated 27 days ago, I am not sure if we still need to change the code in Communication.cs.

Just replace the Json file in Unity3D/Plugin with the AOT Json file.

TomSevestre, 2019-07-25:
> > The Communication.cs file is in ros-sharp/Libraries/RosBridgeClient . However it is updated 27 days ago, I am not sure if we still need to change the code in Communication.cs.
> 
> Just replace the Json file in Unity3D/Plugin with the AOT Json file.

I already did that, but if I'm right I need to download the RosBridgeClient which we can find here in the git, but I don't know what I need to do with it

Scarf195, 2019-07-25:
Download the package from here, after copy the Unity3D folder in your Unity project, you don't need the Libraries and ROS folder, since there are some .dll in Unity3D folder.

TomSevestre, 2019-07-25:
> Download the package from here, after copy the Unity3D folder in your Unity project, you don't need the Libraries and ROS folder, since there are some .dll in Unity3D folder.

Ok so I did what you told me to do and now I have these type of error when I build my Android Application :
Assets\RosSharp\Scripts\RosBridgeClient\ActionHandling\FibonacciActionClientComponent.cs(134,48): error CS0246: The type or namespace name 'Editor' could not be found (are you missing a using directive or an assembly reference?)

Scarf195, 2019-07-25:
Ok, I think you have to add only the RosSharp folder from Unity3D/Asset in your Asset folder of your Unity project.

TomSevestre, 2019-07-25:
> Ok, I think you have to add only the RosSharp folder from Unity3D/Asset in your Asset folder of your Unity project.

I already did that, and I moved the file that made these errors and I build my Application, but it still don't want to subscribe to the topic. But it work when I launch the application on Unity. So I don't know where can the problem come from

Scarf195, 2019-07-25:
Send me your code for the subscribing...maybe I can try if Unity gives me the same errors

TomSevestre, 2019-07-25:
> Send me your code for the subscribing...maybe I can try if Unity gives me the same errors

`using UnityEngine;
using UnityEngine.UI;

namespace RosSharp.RosBridgeClient
{
    [RequireComponent(typeof(RosConnector))]
    public class ImageSubscriber : Subscriber<MessageTypes.Sensor.CompressedImage>
    {

        public RawImage background;

        private Texture2D texture2D;
        private byte[] imageData;
        private bool isMessageReceived;

        protected override void Start()
        {
            base.Start();
            imageData = new byte[20000];
            texture2D = new Texture2D(Screen.width, Screen.height, TextureFormat.RGB24, false);
            Debug.Log("Connected to the camera");

        }
        private void Update()
        {
            if (isMessageReceived)
            {
                ProcessMessage();
            }
        }

        protected override void ReceiveMessage(MessageTypes.Sensor.CompressedImage image)
        {

            imageData = image.data;
            isMessageReceived = true;
            //Debug.Log("Image received");
        }

        private void ProcessMessage()
        {

            texture2D.LoadImage(imageData);
            texture2D.Apply();

            background.texture = texture2D;
            isMessageReceived = false;
        }

    }
}`

TomSevestre, 2019-07-26:
Ok so now I have access to the Debug Log of my Android Application and I see that I have a NullReferenceException : Object reference not set to an instance of an object at RosSharp.RosBridgeClient.ImageSubscriber.Start()

cerenoozturk, 2019-07-26:
I am using Pose Stamped Publisher and  I have the similar problem: 

**NullReferenceException: Object reference not set to an instance of an object.
at RosSharp.RosBridgeClient.PoseStampedPublisher.UpdateMessage () [0x00000] in <00000000000000000000000000000000>:0**

itabr, 2019-08-02:
I have the same issue and I am using the ros-sharp from the unity asset store is there a problem with that package?

berkayalpcakal, 2020-04-15:
Hello everyone. the problem of not subscribing/publishing messages although seemingly working ROS connection was a common problem when you try to build an app for Android devices as well. You can refer to the issue #287 and try again with the settings mentioned. 

As a summary, there has been an improvement in ROS# with #292 which adds a new Json serialization tool. The problem was that Newtonsoft Json serialization does not work for Android devices. It is possible that it also does not work with iOS devices as well. 

I suggest you to pull the recent changes in ROS#, and try out the communication with the settings performed in #287. I would like to try it by myself, but I am not able to build and deploy apps to iOS devices.

Please share your experiences here.

zfquek, 2020-11-30:
Hi, I am using the latest version of ROS# and using Unity version 2019.4.12 to build an Android application. The app is connected to ROS bridge but I am still unable to publish anything to ROS. I have tried out the communication setting as specified in #287 but to no avail.

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff