siemens/ros-sharp #238: URDF Transfer from Unity to ROS without Robot Vizualization

CemAdiyaman, 2019-08-05:
* My OS is: Windows 10 Pro (and Ubuntu 16.04)
* My Unity Version is: 2018.3.0f2
* My ROS Distribution is: Kinetic
* My Build Target Platform is: Windows

**Here is my bug description:**
When I want to visualize the transferred robot by running the command 
```
$ roslaunch file_server visualize_robot.launch model:=$(find <package name>)/robot_description.urdf 
```
with my created package (of course did i changed the name in the command line and deleted the "<" and ">" signs), ROS tells me it can`t find the package, although it was created in the target directory.

Perform the following steps reproduce the bug:
I followed the Wiki step by step and before I started I also deactivated the Hyper-V feature of Windows, so there is no conflict with Oracle VM. Every necessary ROS package was installed without any problems and before every new ROS command I sourced (bashrc). The connecting from Windows and Ubuntu is also working fine. I tried several Versions of Ubuntu, ROS, Unity and Oracle VM, but in all cases the same result was given.

dhrosquete, 2019-08-19:
I have the same Issue under same circunstances

fl0rest, 2019-09-10:
Hi!

I ran into the same problem. Getting the path to the package with "find" doesnt work. The easy workaround I used is to leave out "find" and insert the path manually like this:

`$ roslaunch file_server visualize_robot.launch model:=<relative_path_to_packet>/robot_description.urdf`

In my case (from catkin_ws folder) it was:
`$ roslaunch file_server visualize_robot.launch model:=src/ttbot/robot_description.urdf`

fl0rest, 2019-09-13:
Actually the command doesnt work because "rospack" is missing.
The command in the tutorial should be updated to:

```
$ roslaunch file_server visualize_robot.launch model:=$(rospack find <package name>)/robot_description.urdf
```

MartinBischoff, 2019-09-16:
Thank you @fl0rest . Updated [here](https://github.com/siemens/ros-sharp/wiki/User_App_ROS_TransferURDFToROS).

closed 2019-09-16 by MartinBischoff