siemens/ros-sharp #241: MessageTokenizer crash on Windows style end of line.

Merichbier, 2019-08-08:
* My OS is: Windows 10
* My Unity Version is: Unity 2019.2.0f1
* My ROS Distribution is: not relevant
* My Build Target Platform is: not relevant

**Here is my bug description:**
When trying to auto generate classes for Message/Service/Action, the tokenizer fail.
I tried the unity tool and the console tool, both fail.

Perform the following steps reproduce the bug:
- download [moveit messages definitions](https://github.com/ros-planning/moveit_msgs)
- try to autogenerate any files from the previous git repo
- Failed with message `Invalid character in identifier:  `

I also tried with the following simple msg file:
```
string first_name
string last_name
uint8 age
uint32 score
```
Behavior is the same

Observed results:
The character is actually a windows style line feed `\r`

Expected results:
The file should be able to be converted into a C# class

Relevant Code:
`MessageTokenizer.cs`
`line 119`
``` C#
if (reader.Peek() == '\n')
```
`line 315`
``` C#
while (reader.Peek() != ' ' && reader.Peek() != '\n' && reader.Peek() != '=' && !reader.EndOfStream)
```

Those two lines (as example) only check for linux style end of line. You should create a function that control if the character is an end of line, checking both linux and Windows

sye8, 2019-08-08:
- Couldn't reproduce issue from directly clones MoveIt messages
  - Manually edited with Notepad
- Attempted fix at [my fork](https://github.com/sye8/ros-sharp). Please test.

Thank you!

P.S: Consider getting [Sublime](https://www.sublimetext.com) or [Atom](https://atom.io/). They are awesome

MartinBischoff, 2019-08-12:
resolved in commit #244

closed 2019-08-12 by MartinBischoff

Merichbier, 2019-08-12:
Sorry didn't had time to check, but it should work thanks !