siemens/ros-sharp #243: Message generation fails on string constants

PymZoR, 2019-08-09:
Generating message from this message :
```
string TEST_CONSTANT = test
```

produces the following output :
```c#
/* 
 * This message is auto generated by ROS#. Please DO NOT modify.
 * Note:
 * - Comments from the original code will be written in their own line 
 * - Variable sized arrays will be initialized to array of size 0 
 * Please report any issues at 
 * <https://github.com/siemens/ros-sharp> 
 */

using Newtonsoft.Json;

namespace RosSharp.RosBridgeClient.MessageTypes.VbCom
{
    public class Test : Message
    {
        [JsonIgnore]
        public const string RosMessageName = "VbCom/Test";

        public const string TEST_CONSTANT =  =  test;

        public Test()
        {
        }

        public Test()
        {
        }
    }
}

```

Line 18 should be :
```c#
public const string TEST_CONSTANT = "test";
```

sye8, 2019-08-12:
- Problem reproduced
- Issue located
- Fix should be coming up with the new pull request

Thanks!

MartinBischoff, 2019-08-13:
done in commit #245

closed 2019-08-13 by MartinBischoff