siemens/ros-sharp #248: Citing ROS# for academic work

Rhybo, 2019-08-19:
How should I properly cite ROS# for academic work?  I'm using this package in my thesis and I want to ensure appropriate attribution is given (IEEE format specifically -- a BibTeX citation would be sensational).  I apologize in advance for clobbering the issues forum, but I figured someone else will probably want (or already has) use for it.

MartinBischoff, 2019-08-20:
Hi @Rhybo ,
For citing e.g. release [ROS# Version 1.5](https://github.com/siemens/ros-sharp/releases/tag/v1.5) I'd recommend something like:
```
@misc{bischoffm_2019_06,
    author       = {Martin Bischoff},
    title        = {{ROS#}},
    month        = jun,
    year         = 2019,
    version      = {1.5},
    publisher    = {GitHub},
    url          = {https://github.com/siemens/ros-sharp/releases/tag/v1.5}
    }
```
Improvement suggestions are welcome!

Rhybo, 2019-08-21:
Thank you @MartinBischoff, that is perfect!  This satisfies my issue so I will close the topic.

closed 2019-08-21 by Rhybo

DeathByCh0c0, 2022-03-20:
Very minor point: to include the \# before the #.

As such:
@misc{bischoffm_2019_06,
    author       = {Martin Bischoff},
    title        = {{ROS\#}},
    month        = jun,
    year         = 2019,
    version      = {1.5},
    publisher    = {GitHub},
    url          = {https://github.com/siemens/ros-sharp/releases/tag/v1.5}
    }