siemens/ros-sharp #249: RosBridgeClient dotnet standard fork

ihs-systems, 2019-08-21:
Can someone make dotnet standard / core fork of RosBridgeClient and write some tutorial how to do this? I tryed with websocket standard, but is to old and realy don't know how to do this...

MartinBischoff, 2020-04-21:
Sorry for the very late reply @ihs-systems .
I guess you have forgotten abut this issue a long time ago?

You have to re-build the visual studio solution [here](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosSharp.sln) for the framework / VS version of your choice.

The ROS# source code doesn't include anything special and should be compatible with many frameworks. The dependencies could be problematic. We now have [ISerializer](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Serializers/ISerializer.cs) and [IProtocol](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Protocols/IProtocol.cs) interfaces and ROS# has different variants for Serialization and Websocket commuication.
In case some variants are not compatible with your framework, just delete them. If none of the ones provided works, you can add your own by implementing the corresponding interface.
I hope this late answer still helps anyone... 

Best,
Martin

MartinBischoff, 2020-04-28:
closing due to inactivity.

closed 2020-04-28 by MartinBischoff

markdav-is, 2022-01-05:
I'm looking for a C# ROS client for .Net core.  I see lots of unity/windows (framework) ROS examples, but not so much on Linux.  My thought was to go WASM core-hosted with a ROS client on the core-host and then SignalR to the WASM client.  No cloud in this story.