siemens/ros-sharp #250: Problem with image streaming on android

virtualrobotix, 2019-08-23:
Dear Friends,
i'm trying ros# gazebo simulator on my pc , connect on my ros master node and see without problem image_raw topic on my pc . If i recompile the same app and android and the demo start but on the plane i don't see the raster of my image_raw . 
I'm using the ROS# in asset store and use 2019.3 as version of unity . 
Do you have some suggestion about the configuration of android build for solve my problem . 
Thanks 
Roberto

virtualrobotix, 2019-08-23:
Hi did some debug on android and on pc ... in imagesubscription.cpp the breakpoint 

        protected override void ReceiveMessage(Messages.Sensor.CompressedImage compressedImage)
        {
            imageData = compressedImage.data; // Breakpoint is here 
            isMessageReceived = true;
        }

don't stop .
So could be that the message don't be parsed in right way .

berkayalpcakal, 2020-04-15:
Hello @virtualrobotix,

Could you please refer to #287 and try it again? There has been some changes in ROS#. 
Feel free to open a new issue using the issue template by clearly stating the steps to reproduce the issue.

closed 2020-04-15 by berkayalpcakal