siemens/ros-sharp #251: Websocketclosed error when subscribing to /tf

bomarali, 2019-08-27:
I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I found a bug! ##
I am using the latest ROS# version available here on the master branch.
I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Ubuntu 16.04 with 4.12 rt kernel (server), Windows 10 (client), separate PCs
* My Unity Version is: 2019.2
* My ROS Distribution is: Kinetic
* My Build Target Platform is: Unity Editor

**Here is my bug description:**
The rosbridge throws a "WebSocketClosedError: tried to write to a closed websocket" warning every second when client subscribes to a non-empty /tf topic.

Perform the following steps reproduce the bug:
I use this code to subscribe to the tf:

using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using System;

namespace RosSharp.RosBridgeClient
{
    public class tf_message_subscriber : Subscriber<MessageTypes.Tf2.TFMessage>
    {
        private bool isMessageReceived;
        public Dictionary<string, Transform> tfs;
        public Transform marker;
        public Transform robot;
        public Transform kinect2;
        private Quaternion _q;
        private Vector3 _v;
        private string _name;

        protected override void Start()
        {
            base.Start();
            tfs = new Dictionary<string, Transform>();
            tfs.Add("marker", marker);
            tfs.Add("panda_link0", robot);
            tfs.Add("top_kinect2_ir_optical_frame", kinect2);
        }

        private void Update()
        {
            if (isMessageReceived) ProcessMessage();
        }

        protected override void ReceiveMessage(MessageTypes.Tf2.TFMessage message)
        {
            isMessageReceived = true;
            if (tfs.ContainsKey(message.transforms[0].child_frame_id))
            {
                _v = new Vector3((float)message.transforms[0].transform.translation.x,
                    (float)message.transforms[0].transform.translation.y,
                    (float)message.transforms[0].transform.translation.z);
                _q = new Quaternion((float)message.transforms[0].transform.rotation.x,
                    (float)message.transforms[0].transform.rotation.y,
                    (float)message.transforms[0].transform.rotation.z,
                    (float)message.transforms[0].transform.rotation.w);
                _name = message.transforms[0].child_frame_id;
            }
        }

        private void ProcessMessage()
        {
            tfs[_name].localPosition = _v.Ros2Unity();
            tfs[_name].localRotation = _q.Ros2Unity();
            //tfs[_name].localPosition = _v;
            //tfs[_name].localRotation = _q;
        }
    }
}

Observed results:
The rosbridge throws a "WebSocketClosedError: tried to write to a closed websocket" warning every second when client subscribes to a non-empty /tf topic. If there are no /tf messages on server the warning does not appear.

Expected results:
There should be no warning.

bomarali, 2019-08-29:
A follow-up: i get similar behaviour when I subscribe to other messages that are not originally included in ros-sharp. I have made a subscriber for the octomap_binary message, it also causes the warning. 

Slightly offtopic but also an issue: the client sometimes fails to subscribe, giving this error in Unity:
NullReferenceException: Object reference not set to an instance of an object
System.Collections.Generic.Dictionary`2[TKey,TValue].TryInsert (TKey key, TValue value, System.Collections.Generic.InsertionBehavior behavior) (at <a8ed250850854b439cedc18931a314fe>:0)
System.Collections.Generic.Dictionary`2[TKey,TValue].Add (TKey key, TValue value) (at <a8ed250850854b439cedc18931a314fe>:0)
RosSharp.RosBridgeClient.RosSocket.Subscribe[T] (System.String topic, RosSharp.RosBridgeClient.SubscriptionHandler`1[T] subscriptionHandler, System.Int32 throttle_rate, System.Int32 queue_length, System.Int32 fragment_size, System.String compression) (at <b3ed88deda0a4787b6a85cac2c3d4701>:0)
RosSharp.RosBridgeClient.Subscriber`1[T].Subscribe () (at Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/Subscriber.cs:42)
System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) (at <a8ed250850854b439cedc18931a314fe>:0)
System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <a8ed250850854b439cedc18931a314fe>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) (at <a8ed250850854b439cedc18931a314fe>:0)
System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) (at <a8ed250850854b439cedc18931a314fe>:0)
System.Threading.ThreadHelper.ThreadStart () (at <a8ed250850854b439cedc18931a314fe>:0)
UnityEngine.UnhandledExceptionHandler:<RegisterUECatcher>m__0(Object, UnhandledExceptionEventArgs)
On server side I can see that the client fails to subscribe to a one of four topics I am interested in: octomap_binary, tf, 2 poses.

berkayalpcakal, 2020-03-25:
Hey @bomarali and @m-talha.
Any updates on the issue? There have been some changes in the repo, so are the steps to reproduce the issue still valid? I would like to reproduce the issue, and try to resolve it.

bomarali, 2020-03-26:
Frankly I do not remember how I resolved this issue. I have dropped the tf subscriber. I had no problems with creating new message types. One thing that might be important is to have a scheduler that activates publishers and subscribers one by one with little pauses in between - I have 10+ subscribers/publishers running in my setup, so throwing them simultaneously at the websocket tends to be problematic. Other than that I do not have any updates.

closed 2020-03-26 by bomarali