siemens/ros-sharp #254: Can we use this package for WebGL building?

mugi-systems, 2019-09-09:
I use this package for my programs and tried to build them for WebGL. Build succeeded, but alert dialog was displayed and could not be executed.

So, as a test, I tried to build a scene which contains camera, light and empty object (attaced 'RosConnector.cs' and 'TwistSubscriber.cs'). But, the result was the same as my programs.
Also, I tried to attach 'RosConnector.cs' only. The build was successful and executed without problems. In short, it seems that error is occurred when the subscriber script was attached.

This package could not be used for WebGL? If someone knows any solutions about this issue, please teach me.

This is my environment.
- OS:  Mac OSX 10.13.6
- Unity: 2018.4.4f1 personal

This is error scripts displayed in web browser.
```
44770de7-3973-409f-9dde-d701cd3039f8:2 Uncaught abort(72) at Error
   at jsStackTrace (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:15158)
   at Object.stackTrace (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:15329)
   at Object.onAbort (http://localhost:53226/Build/UnityLoader.js:4:9327)
   at abort (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:503679)
   at wasm-function[74972]:4
   at wasm-function[25255]:35
   at wasm-function[26807]:142
   at wasm-function[26808]:70
   at wasm-function[55207]:273
   at wasm-function[61368]:45
   at wasm-function[20409]:17
   at wasm-function[74849]:21
   at Object.dynCall_iiiii (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:477906)
   at invoke_iiiii (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:358456)
   at wasm-function[72688]:172
   at wasm-function[73107]:36
   at wasm-function[2667]:64
   at wasm-function[2666]:109
   at wasm-function[5242]:2687
   at wasm-function[5237]:150
   at wasm-function[7850]:199
   at wasm-function[5721]:936
   at wasm-function[8632]:33
   at wasm-function[7945]:154
   at wasm-function[7945]:175
   at wasm-function[7939]:293
   at wasm-function[7934]:280
   at wasm-function[74870]:12
   at Object.dynCall_v (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:488305)
   at browserIterationFunc (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:125778)
   at Object.runIter (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:128851)
   at Browser_mainLoop_runner (blob:http://localhost:53226/44770de7-3973-409f-9dde-d701cd3039f8:2:127313)
```

berkayalpcakal, 2020-04-21:
Hello @mugi-systems, do you have any update on this issue?

First let us clarify the issue a bit. Do you observe that you cannot subscribe/publish to a topic even though it seems that the application connects to ROS successfully? 

I do not have any experience with WebGL but if this is the case, then it seems the problem is similar to other issues regarding different platforms such as Android and iOS, where the problem is mainly because of the incompatibility of Newtonsoft Json serialization tool. 

I suggest that you first pull this repository as there have been an improvement in ROS# with #292 which adds a new Json serialization tool. Then on Unity side, simply set the `Serializer-Protocol` settings to `Microsoft - WebSocketSharp` on the inspector window of RosConnector.cs.

Please post your experiences here as it would help the community.

fredrikbaberg, 2020-10-06:
I'm having a similar issue (I think). I'm able to build for WebGL, but when I run it I'm getting an error message similar to what is shown above.
I already had `Microsoft - WebSocketSharp` set, and have also tried to change .NET version.

I'm on Ubuntu 20.04, Unity 2019.4.11f1.

berkayalpcakal, 2020-10-19:
I also reproduced this issue. Error messages do not seem to be self explanatory, and I am not really a WebGL guy. 

However here is what I found after searching a bit on the Unity documentation; 

> Not all features of Unity are available in WebGL builds, mostly due to constraints of the platform. Specifically:
> * Threads are not supported due to the lack of threading supporting in JavaScript. This applies to both Unity’s internal use of threads to speed up performance, and to the use of threads in script code and managed dlls. Essentially, anything in the System.Threading namespace is not supported. 

ROS# heavily depends on threading due to concurrent handling of the message flow. At this point it seems that it is not possible to use ROS# for WebGL buildings. 

There is another library [roslibjs](http://wiki.ros.org/roslibjs) to communicate between a web browser and ROS over rosbridge websocket server.

closed 2020-10-19 by berkayalpcakal