siemens/ros-sharp #255: Websocket is closed for non-UTF8 binary messages

limond, 2019-09-12:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch. **But I believe the problem to be universal** as it appears in my netstandard build and the last official unity release.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Debian GNU/Linux bullseye/sid
* My Unity Version is: 2019.3.0a10
* My ROS Distribution is: Kinetic
* My Build Target Platform is: Debian/netstandard + Unity(Debian)/mono

**Here is my bug description:**

Most BSON-serialized messages (`RosSocket` created with `RosSocket.SerializerEnum.BSON`) close the WebSocket if the protocol is `WebSocketNET`. The [`sendAsync`](https://github.com/siemens/ros-sharp/blob/edfc69c1ad9d3cf5661737dd0724719c585b3ce8/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs#L99) method is invoked with `WebSocketMessageType.Text` as Opcode for the WebSocket message.

This indicates a message with a payload that can be interpreted as an UTF8 string. This is not the case for BSON-encoded messages. Most messages force the rosbridge server to close its WebSocket as required. [RFC](https://tools.ietf.org/html/rfc6455#section-8.1):

> When an endpoint is to interpret a byte stream as UTF-8 but finds
   that the byte stream is not, in fact, a valid UTF-8 stream, that
   endpoint MUST _Fail the WebSocket Connection_.

If the `WebSocketSharp` backend is used, this does not appear. The binary opcode [is used](https://github.com/sta/websocket-sharp/blob/beb3a149761ff4aa144037ed2a40725cca02306f/websocket-sharp/WebSocket.cs#L3573) for JSON and BSON encoded messages.

Perform the following steps reproduce the bug:

Send a BSON serialized message (an empty Sensor.Imu message does it for me, most Std messages do not produce invalid UTF8 as it seems. A long String message works too) to a rosbridge server configured with `bson_only_mode:=True`

Observed results:

The WebSocket is closed by the rosbridge server:

```
ros-recorder_1     | Unhandled Exception: System.Net.WebSockets.WebSocketException: The remote party closed the WebSocket connection without completing the close handshake. ---> System.IO.IOException: Unable to read data from the transport connection: Connection reset by peer. ---> System.Net.Sockets.SocketException: Connection reset by peer
ros-recorder_1     |    --- End of inner exception stack trace ---
ros-recorder_1     |    at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error)
ros-recorder_1     |    at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.GetResult(Int16 token)
ros-recorder_1     |    at System.Net.Http.HttpConnection.ReadBufferedAsyncCore(Memory`1 destination)
ros-recorder_1     |    at System.Net.Http.HttpConnection.RawConnectionStream.ReadAsync(Memory`1 buffer, CancellationToken cancellationToken)
ros-recorder_1     |    at System.Net.WebSockets.ManagedWebSocket.EnsureBufferContainsAsync(Int32 minimumRequiredBytes, CancellationToken cancellationToken, Boolean throwOnPrematureClosure)
ros-recorder_1     |    at System.Net.WebSockets.ManagedWebSocket.ReceiveAsyncPrivate[TWebSocketReceiveResultGetter,TWebSocketReceiveResult](Memory`1 payloadBuffer, CancellationToken cancellationToken, TWebSocketReceiveResultGetter resultGetter)
ros-recorder_1     |    --- End of inner exception stack trace ---
ros-recorder_1     |    at System.Net.WebSockets.ManagedWebSocket.ReceiveAsyncPrivate[TWebSocketReceiveResultGetter,TWebSocketReceiveResult](Memory`1 payloadBuffer, CancellationToken cancellationToken, TWebSocketReceiveResultGetter resultGetter)
ros-recorder_1     |    at RosSharp.RosBridgeClient.Protocols.WebSocketNetProtocol.StartListen() in /root/dev/ros-sharp-common/ros-sharp/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs:line 113
ros-recorder_1     |    at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
ros-recorder_1     | --- End of stack trace from previous location where exception was thrown ---
ros-recorder_1     |    at System.Threading.ThreadPoolWorkQueue.Dispatch()
```
Expected results:

The WebSocket is not closed by the rosbridge server.

Relevant Code:
```C#
RosConnector.ConnectToRos(Protocol.WebSocketSharp, info.serverURL,
                    onROSConnected, onROSDisconnected, RosSocket.SerializerEnum.BSON);
//...
var id = socket.Advertise<RosSharp.RosBridgeClient.MessageTypes.Sensor.Imu>("/device/imu");
socket.Publish(id, new RosSharp.RosBridgeClient.MessageTypes.Sensor.Imu());
```

MartinBischoff, 2019-09-12:
Does `WebSocketMessageType.Binary` instead of `WebSocketMessageType.Text` solve this issue?

MartinBischoff, 2019-09-13:
Thanks for your evaluations and reports @limond . Would you please verify if also JSON messages can be transported with `WebSocketMessageType.Binary` option? If  `WebSocketMessageType.Text` is required for JSON, we will have to distinguish between JSON and BSON here. @akirayou fyi

limond, 2019-09-13:
I will try and confirm this in a build of the master branch in Windows with an actual Net Framework over the weekend.

My guess is that it does not matter and it is possible to use `WebSocketMessageType.Binary` for JSON messages as well.

The follwing is my understanding from reading the source code/documentation of related code:
- If you use the WebSocketSharp implementation, the Binary Opcode is set for JSON Messages and they are received correctly
- The current websocket implementation in rosbridge (tornado) calls its [`on_message`](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler.on_message) handler either with a string or bytes [depending on the opcode](https://github.com/tornadoweb/tornado/issues/2031#issuecomment-298805354)
  - rosbridge [then](https://github.com/RobotWebTools/rosbridge_suite/blob/55b88174fe6da6202d0522aba6c5a763e4ddf88b/rosbridge_server/src/rosbridge_server/websocket_handler.py#L122) calls [`json_loads`](https://docs.python.org/3/library/json.html#json.loads) with the payload (if the server is configured without `bson_only_mode`). The documantation explicitly allows this to be bytes if they represent a UTF-(8/16/32) string.

limond, 2019-09-16:
I tested this under Windows 10 (VS2017/Net Framework 4.6) against a kinetic rosbridge [BSON and JSON server](https://gist.github.com/limond/b1b99e7ef1d1c77017adf5abf832feb7).

All tests in `RosSocketTests.cs` are successfull when using `WebSocketMessageType.Binary` regardless of whether I use `SerializerEnum.JSON` or `SerializerEnum.BSON` and follow the instructions in this test class's comments.

I can also confirm that this bug can be replicated in this configuration when using `WebSocketMessageType.Text` as soon as the payload includes characters that can not be interpreted as UTF8.

MartinBischoff, 2019-09-16:
Thank you @limond for your tests and reports.
Issue is solved in commit 2891dd4.

closed 2019-09-16 by MartinBischoff