siemens/ros-sharp #257: Can we get a step-by-step of how to make ROS# work in Ubuntu?

HCurrence, 2019-09-23:
* [ *] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ *] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I have a question! ##

* [* ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I saw that someone found a way to make ROS# work in the Unity3D experimental build for Ubuntu in #74. I am working on the same thing. While I know I am not alone in this, I can't seem to find any documentation regarding how to integrate ROS# into Unity3D. Is there any available?

Regarding my process, I have the ROS# package in the assets folder of my Unity project and the scripting settings set to .NET 4.x. Unity acknowledges the existence of the files but the plugin doesn't load in. 

I'm currently using Ubuntu 16.04, Unity 2019.2.6f1, and ROS Kinetic.
Thanks!

HCurrence, 2019-09-24:
Thanks for the quick response! I'll try replacing the dlls, and see how that works. If I find anything, I'll be sure to document my findings.

PymZoR, 2019-09-26:
@HCurrence I'm working with ROS melodic on ubuntu 18.04, and Unity 2019.3.0a8. ROS# is working fine with the dlls. I have tried to use the sources instead for tweaking, but without success, as there are errors on namespaces conflicts (multiple definition of classes). I haven't digged further.

MartinBischoff, 2019-10-08:
We fixed the naming problems in commit https://github.com/siemens/ros-sharp/commit/f8b6af52cfc4a43e7bf812afeeb0b83e11b645f6
The library sources now compile in Unity on a Windows system.

If there are any additional steps necessary to get ROS# running on an Ubuntu system please let me know, I'd be happy to include it in the Wiki.

HCurrence, 2019-10-13:
Hey guys, I apologize for the lack of response. Life.

Before I tried replacing the dlls in my project; but, I have pulled from commit f8b6af5 and reset my workspace accordingly. Unfortunately, no such luck there. Despite this, I attempted to replace the dlls manually with the same result.