siemens/ros-sharp #261: ActionClient creates an extra ros connection?

bomarali, 2019-10-28:
[Yes] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
[Yes] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

[Yes] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
The fibbonaci action client creates the ros connection (presumably in the actionclient class). So if I have both a ros_connector (for other publishers) and fibbonaci action client, I end up with two ros_connections (I get two clients on the rosbridge websocket server). So far it looks to me that the second connection is redundant, isn't it? I would like to clarify if it is actually necessary.

MartinBischoff, 2019-10-29:
Hi @bomarali ,
both in [ActionClient](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Actionlib/ActionClient.cs) and [ActionServer](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Actionlib/ActionServer.cs) a new RosSocket is generated.
I agree that we should change the constructor and pass an existing RosSocket instance instead.

If I remember right, this implementation comes from a problem we had during development, where we tried to advertise and subscribe action topics before the corresponding RosSocket connection had been opened.

bomarali, 2019-10-29:
I see, thank you for the reply. 

> If I remember right, this implementation comes from a problem we had during development, where we tried to advertise and subscribe action topics before the corresponding RosSocket connection had been opened.

I am surprised that this is a problem. The ActionClient (I did not try the server yet) is just a 1 ros_socket, 3 subscribers and 2 publishers. They work fine on their own, at least it is possible to do:
`if (!rosConnector.IsConnected.WaitOne(SecondsTimeout * 1000)) `
I will try to modify the constructor for the ActionClient.

closed 2019-10-29 by bomarali