siemens/ros-sharp #264: Cannot inherit from publisher or subscriber (and maybe other classes) because theyr are internal and not public

flonou, 2019-10-31:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [ x I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: windows 10
* My Unity Version is: 2018.4
* My ROS Distribution is: melodic
* My Build Target Platform is: x64

**Here is my bug description:**
The documentation at https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes describes how to create a publisher or a subscriber  but as the classes are now in an assembly and internal, this can't be done anymore

Relevant Code:
```
  public class FloatPublisher : Publisher<MessageTypes.Std.Float32>
```
error CS0122: 'Publisher<T>' is inaccessible due to its protection level

flonou, 2019-10-31:
Found out that these class are now named UnityPublisher, .... that's why inheritance couldn't work

flonou, 2019-10-31:
Then it's about updating the wiki

flonou, 2019-10-31:
Hi,
The last two code sections at https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes still refer to Subscriber and publisher

MartinBischoff, 2019-10-31:
I just changed these two. Thanks for reporting!
Please let me know if there is anything else;
otherwise, please close this issue.

closed 2019-11-11 by MartinBischoff