siemens/ros-sharp #267: their is no file_server.launch file in file server

frotein, 2019-11-08:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2019-11-11:
Hi @frotein ,

why do you need a separate `file_server.launch` file? 

`file_server` is launched together with `rosbridge_websocket` in [ros_sharp_communication](https://github.com/siemens/ros-sharp/blob/master/ROS/file_server/launch/ros_sharp_communication.launch).

In the unusal case that you need the `file_server` service without `rosbridge` you can go with `rosrun file_server file_server`.

gonzalocasas, 2019-11-15:
Well, is not such an uncommon case to want to use your file server on its own. We use it without ros-sharp, so we detached and forked it: https://github.com/gramaziokohler/ros_file_server

MartinBischoff, 2019-11-18:
Just asking out of curiosity: What are you using the file_server for? Which connections do you use it with, if not `rosbridge_suite` websocket server?

gonzalocasas, 2019-11-18:
We definitely use it with `rosbridge_suite`, what we don't use it with is ros-sharp. That's why we detached it and made it a standalone project. If you'd split it on your official repo, we would drop our fork and use yours instead.

MartinBischoff, 2019-11-29:
Ok, so you are fine with the [ros_sharp_communication launch file](https://github.com/siemens/ros-sharp/blob/master/ROS/file_server/launch/ros_sharp_communication.launch) launch file, but your issue is that the file_server is on the same repo as the rest of ROS#?

I agree that it is quite cumbersome to download a whole repo when you only need individual files. Please do feel free to split the ROS# code into individual repos as you need it for your application.

The advantages of keeping everything in one repo here are:
* We can always ensure consistency between ROS and .NET code; no discussion about which file_server version goes with that Unity3D version.
* Beginners immediately see what is needed on the ROS side to get things running and do not have to download code from different places.

closed 2019-11-29 by MartinBischoff