siemens/ros-sharp #268: Get message ID name

arBalasquide, 2019-11-11:
* [X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I'm trying to make a simple UI to toggle a topic on and off for debugging purposes. I have all my messages in an array and would like to be able to do something like messages[index].getID()

So I can later publish and unadvertise in an abstract manner, since I would also be adding future messages to the UI, I need it to be able to easily add it to an array and be done.

Any sort of guidance would be appreciated.

MartinBischoff, 2019-11-11:
Hi @arBalasquide , [rosapi](https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosapi) will help you:

> rosapi provides service calls for getting meta-information related to ROS like topic lists as well as interacting with the Parameter Server.

Launching a `rosbridge_server` includes `rosapi` already.

On the Unity side you can call those services and build your message array from the results.

arBalasquide, 2019-11-11:
@MartinBischoff thanks for the reply. 

I'm not really sure on how to call rosapi services from Unity using ros#
Are there any examples or documentation for that in this repo?

arBalasquide, 2019-11-11:
Ah, found 
[(https://github.com/siemens/ros-sharp/blob/711396cf78578cc95aca49d2f6443ed406440376/Libraries/RosBridgeClientTest/RosSocketConsoleExample.cs)]

I think I can work it out from this. Thanks for the guidance I'll close the issue now.

closed 2019-11-11 by arBalasquide