siemens/ros-sharp #273: Latched Topics Support

davidbsp, 2019-12-13:
Hi there, my question is simple: is there support for latched publishers?
My use case is simple: I would like to advertise a topic from the unity side, which will only published when a new subscriber from the ROS side is listening.

Official info on ROS latched connections [here].(http://wiki.ros.org/roscpp/Overview/Publishers%20and%20Subscribers#Publisher_Options)

MartinBischoff, 2019-12-13:
Ideally, `rosbrige_suite` should provide a latching method, not ROS#.
And it already does, however for ROS-internal topics only, but not __yet__ for external topics coming from e.g. Unity or other .NET applications via ROS#.
See Section 4.3 in the [rosbridge Protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md).

We did not do anything in that direction and I think I wouldn't like a latching functionality in ROS# in addition to what is already provided by `rosbridge_server`.

I see two workarounds/solutions for you:

1. Implement a latching functionality in Unity yourself by listening to [rosapi/Subscribers](http://docs.ros.org/jade/api/rosapi/html/srv/Subscribers.html) and publish based on the subscribers existing for each topic.

2. Have you thought about using a service instead? Or a service/action that initializes and terminates the topic, if you want to receive a message repetitively from a certain point of time on. You colud even initialize and terminate complete virtual robotic systems publishing and subscribing a bunch of topics.

davidbsp, 2019-12-13:
Hi @MartinBischoff, thank you for your prompt answer.
You're right, This should be supported by rosbrdige and not ROS# itself.
Our plan B is to use a service instead, but we'll look into the latching funcionality in Unity ourselves.
Thanks for your feedback!

closed 2019-12-13 by MartinBischoff