siemens/ros-sharp #274: Need a basic tutorial on how to publish depth images from a Unity3D camera on a ROS topic.

BhavyanshM, 2019-12-20:
* [X ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I have been trying to find basic documentation on how to generate and publish a depth camera stream from Unity to ROS. I am a PhD student currently working on visual SLAM algorithms and being able to simulate an RGB-D camera on Unity and being able to use it in ROS would be a really really important availability for me. I would sincerely request help and guidance on this issue. Thank you a lot. 

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2019-12-20:
Hi @BhavyanshM ,
in the ROS# framework there exists no depth camera model that you can use out of the box.

You will have to generate the 3d point cloud data in Unity according to your application specific needs. Here are two suggestions:
* use [Physics.raycast](https://docs.unity3d.com/ScriptReference/Physics.Raycast.html)
* use one or two Camera images and use a SLAM method of your choice
* you find other Unity point cloud generation approaches in the net (e.g. [this](https://unitylist.com/p/fa0/pointcloud-Unity-example))

Then you can put this data into into a point cloud message (e.g. a [PointCloud2](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/PointCloud2.html)) and send it to your ROS system using ROS# and rosbridge_suite.

Likely there are others working someting similar. Please feel free to use this issue to discuss your approaches.

MartinBischoff, 2020-04-21:
As depth images are very application specific, we don't want to include it into ROS#.
Since there are no discussions about applications ongoing, I'm closing this issue.

closed 2020-04-21 by MartinBischoff