siemens/ros-sharp #275: Topic conflict for /odom published by Pose Stamped Publisher

Konye666, 2020-01-02:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

berkayalpcakal, 2020-03-25:
Hello @Konye666, do you have any updates? 
I could not reproduce the issue, so could you please use the latest version of ROS# and write down the steps to reproduce the issue?

berkayalpcakal, 2020-03-30:
Closed due to inactivity.

closed 2020-03-30 by berkayalpcakal

amelia413, 2022-11-07:
I'm facing the same issue! 
Unity : 2019.4.40f1
Ros: Noetic
Ros# latest version