siemens/ros-sharp #277: Best way to use RosConnectors when there are several game objects

karthikm-0, 2020-01-08:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
Thank you very much for providing such a great package to connect ROS and Unity! I am currently building a simulation where there are are multiple Gazebo objects represented in Unity. Each of these objects subscribes to state updates from Gazebo and sometimes can publish as well (through individual subscribers and publishers for each object). However, I have quickly noticed that scaling this setup to multiple objects (with each of them having their own RosConnector) slows down the application tremendously. The obvious solution for this seems to be to only use one RosConnector and funnel all updates to each individual game object (which will not have RosConnectors). However, I was wondering if anyone would have more insight into what causes this bottleneck and whether there are alternative ways to fix it.

MartinBischoff, 2020-01-09:
Hi @karthikm-0 ,

we designed RosConnector to be used only once per connected ROS system. Otherwise, you will set up multiple Websocket connections between your Unity machine and your ROS machine which costs performance.

You can add as many Publisher and Subscriber components (also of the same type) to the RosConnector as you like and place the corresponding Readers and Writers to any GameObject in your scene.

What is your reason for separating your Publishers and Subscribers to various GameObjects in your scene?

If you want to have individual Publisher/Subscriber GameObject for each of your Gazebo objects (for a better overview?), you could proceed as follows:

1. modify [UnityPublisher](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnityPublisher.cs) and [UnitySubscriber](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnitySubscriber.cs) as follows:
* Delete the line
```
[RequireComponent(typeof(RosConnector))]
```
* Replace the line
```
rosConnector = GetComponent<RosConnector>();
```
by
```
rosConnector = GetComponentInParent<RosConnector>();
```

2. add subfolders to your RosConnector gameobject and place your publishers & subscribers there.

Please keep us informed about the results.

Best,
Martin

karthikm-0, 2020-01-13:
Thank you very much, Dr. Bischoff! Now that I think about it, I completely overlooked the state writers, which perfectly fit my needs. So I don't actually need multiple subs/pubs. But it's nice to know that there is a way to separate them out. Thanks again!

closed 2020-01-13 by karthikm-0