siemens/ros-sharp #278: Hololens 1 connecting on Rosbridge but not communicating both on topics and actions

aioniosfoititis, 2020-01-10:
[image]
[image]
[image]

* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Observed results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---

aioniosfoititis, 2020-01-10:
I'm trying to communicate Hololens 1 with ROS. When I use the unity play everything is occuring as expected.When the project is build although the hololens are connected but there is no no communication,tried both topics and action messages.Both tried in build settings  .NET 2.0 and 4.x as well as VS2017 and VS2019.

## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is:

berkayalpcakal, 2020-01-10:
Hello @aioniosfoititis,

Have you tried the repo from @dwhit? Here is the [link](https://github.com/dwhit/ros-sharp) for UWP supported version for ROS#. Currently I am able to do two way communication of simple messages and action messages between ROS and a Unity application that is deployed to Hololens 1.

aioniosfoititis, 2020-01-10:
@berkayalpcakal Thanks for the quick reply.No I haven't noticed that that exists.Huge relief that there is actually support for that. I will test that  as soon as possible and come again to report if everything works fine.

aioniosfoititis, 2020-01-11:
Tried the modified version that you mentioned,but Im getting the following error when im trying to deploy the app to the hololens :"Failed to connect to device '127.0.0.1' using Universal Authentication".
That didnt happen before as with the default Unity3D Im deploying without a problem.

aioniosfoititis, 2020-01-13:
@berkayalpcakal Still the same result.Connection but no communication.
Is there any special step I should have done in order to make the system connect?
Im inserting Mixed Reality toolkit ,placing Ros Connector and eg fibonacci server and adding action name and ip address.

With the play button everything occurs with success.On the other hand when deployed in the hololens I can only notice a connection to the rosbridge but no subscribing to the fibonacci topics.

berkayalpcakal, 2020-01-15:
Hi again @aioniosfoititis,

Sadly I am not able to reproduce the issues that you have. The only recommendation I can make is that maybe you should check the "Project Settings/Player/Capabilities" ( I am not sure if I remember the path correctly, you can go ahead and check this [link](https://docs.microsoft.com/en-us/windows/mixed-reality/recommended-settings-for-unity), and enable network related fields. Maybe there is an unchecked field which prevents the Unity app that is deployed to Hololens from having  a proper communication to ROS installed PC.

aioniosfoititis, 2020-02-04:
To anyone that listens,
 As for the capabilities I'm using InternetClient, InternetClientServer and privateClientServer.

I also tried to not use the virtual machine and use an actual PC, but the problem still occurs.I can see message "one client connected" but no subscribing to any topics happens.
After that I tried the Export commands of ROS ( ip,hostname ,master uri ) but it didnt real overcome the problem.
Also tried the UWP version of the websocket in the RosConnector .
Anyone that has exei more suggestions please feel free to respond and I will give it a try.

helenol, 2020-02-24:
To revive this a bit, ran into the same issues and was able to get around them by following this thread:
https://github.com/dwhit/ros-sharp/issues/3

Solution was to use @dwhit's fork and replace the Netwonsoft.json dll with the *AOP* one from the asset store here: https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347

aioniosfoititis, 2020-02-24:
I can also confirm the mentioned fix!

closed 2020-02-24 by aioniosfoititis

aoboaten1, 2020-02-25:
@helenol I tried exactly what you said and it does not work for me.

aioniosfoititis, 2020-02-26:
@aoboaten1 Can u name what are u using,unity version ,ros# version, ubuntu version?

aoboaten1, 2020-02-28:
> @aoboaten1 Can u name what are u using,unity version ,ros# version, ubuntu version?
OS: WINDOWS 10
Unity Version : Unity 2019.2.20f1 (64-bit)
ROS Distribution : ROS Kinetic
I use use the @dwhit's fork and followed all instructions

helenol, 2020-02-28:
Try 2019.3.1 for Unity as first debug step. I'm also using Kinetic. (Using
Hololens 2 though).

aoboaten1, 2020-02-28:
> Try 2019.3.1 for Unity as first debug step. I'm also using Kinetic. (Using Hololens 2 though).
Yeah. I have used 2019.3.1 as well but it didnt work. I use hololens.
I went back to an old version of the Ros# I had downloaded and built with .net backend instead of  il2cpp and it worked

aioniosfoititis, 2020-02-28:
@aoboaten1  So if you are using .net backend you have probably used 2018.x version of unity as 2019.x does not support .net anymore ,If I recall correctly.

aoboaten1, 2020-02-28:
@aioniosfoititis Yes. I have about 4 versions on of unity on my laptop. I downloaded the newer version of the ros# and it wouldn't work so I went to my old laptop to use the old ros# i had and used 2018 unity version.

BloodRaine, 2020-07-20:
Hello, I am currently trying to do something like what has been discussed in this thread, however I have been having issues getting it to run properly. For reference I currently have my ROS nodes and topics running on a ubuntu computer connected to my network and running/building my unity in windows. Same thing as above, I get rosbridge connection, but no subcribing to topics. I'm running on windows 10 with Unity 2019.3.14f1 using ROS Melodic.

Thus far I have done the following:
- Imported and set up the MRTK
- Imported @dwhit version of ROS#
- Imported JsonDotNet unity Package
- Renamed initial ROS# Newtonsoft.Json.dll to NewtonsoftR.Json.dll to avoid dual file conflict
- Moved AOT Newtonsoft.Json.dll and attached xml file into RosSharp/Plugins

In the MRTK build window I am using:
-build configuration: release
- platform: x86

I am able to use the build and install method, However as stated previously, I get a rosbridge connection, but it does not subscribe to my image topic, and when run in Unity it works just fine. Any assistance would be greatly appreciated.

berkayalpcakal, 2020-07-20:
Hello @BloodRaine, 
just asking for curiosity, have you tried [.NET Serializer](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Serializers/MicrosoftSerializer.cs) (preferably with WebSocketSharp protocol) instead of Newtonsoft serializer? You can see the corresponding settings in the RosConnector component.

BloodRaine, 2020-07-20:
@berkayalpcakal I have not, Any resources/ways to set that up?

berkayalpcakal, 2020-07-20:
I think the fork by @dwhit is old, so you need to adjust the new changes in the serialization interface (please see #292) to your code. Or, you can adjust the modifications done in that fork to the current ROS#, see my comment [here](https://github.com/siemens/ros-sharp/issues/201#issuecomment-632665786). I hope this helps. I would be glad if you can comment your experiences regarding .NET Serialization tool with HoloLens here.

BloodRaine, 2020-07-20:
Just so that I am making sure I'm understanding you correcltly @berkayalpcakal, I can:
- adjust the new serialization changes to @dwhit 's fork **OR**
- reperform the modifications of the fork on the current ROS# master as a new fork
am I understaning that correctly? If so I will likely go with the second option as that would meke the most sense to me.

BloodRaine, 2020-07-27:
@berkayalpcakal after updating to the latest ros-sharp, and updating subscriber scripts to follow newer syntax for Messages -> MessageTypes and Subscriber -> UnitySubscriber, I get the error "ReflectionTypeLoadException ... etc" is this something that you have seen before?

Jntzko, 2020-07-27:
This could help with the Reflection issue: https://github.com/siemens/ros-sharp/issues/201#issuecomment-646590687
I'm interested in your branch, could you upload it?

BloodRaine, 2020-07-27:
Yeah thats very true, after taking steps that were defined in some of the linked comment threads above, I managed to get around that issue using [this](https://github.com/EricVoll/ros-sharp) fork of ros-sharp, however, I am running into some issues of certain namespaces not being found by the Newtonsoft.Json such as JsonConvert. @Jntzko  once I have a working system I plan to create my own fork/version, unless it would help in debugging what issues I am running into.

BloodRaine, 2020-08-03:
@Jntzko, have you managed to get a working connection with topic subscription/publishing? I think I have my dll's set properly, however, I am having other libraries throw issues. For my project I am using OpenCVForUnity and that seems to be having issues when my dll's are set properly. I haven't yet gone as far as to take ros-sharp and re-remove the reflection code as @EricVoll had previously done.

Jntzko, 2020-08-04:
I didn't have it running yet, I use the current ros-sharp master branch and applied the changes from [dwhits](https://github.com/dwhit/ros-sharp) fork, but I can't connect to topics, even with the json-net-for-unity dll, independent of the serializer. I didn't applied the reflection fix yet, maybe that helps.

EricVoll, 2020-08-04:
Hi guys,
I'll comment in 1-2 hours and give you some links. On mobile atm...
Maybe I can help :)

Jntzko, 2020-08-04:
Thanks a lot for your detailed description, this definitely cleared up some things. [ARbotics](https://github.com/EricVoll/ARbotics) looks very interesting, I didn't new about that work.
Unfortunately, I can start integrating your suggestions in 2 weeks, but I'm pretty sure they will help a lot!

Jntzko, 2020-08-04:
You were right @EricVoll, thanks a lot. I was able to publish to a topic from the HoloLens 2 emulator, I just copied the RosSharp and the plugin folder from the Assets of [ARbotics](https://github.com/EricVoll/ARbotics). I don't have access to the real hardware right now.

My setup right now is:
Unity 2019.3.15f and MRTK 2.4.0

EricVoll, 2020-08-04:
Happy to hear, that everything worked out so far. I exchanged a few messages today with @MartinBischoff who will probably/maybe add a link to the specific fork on the Ros# Readme under "Ros# for HoloLens" (or something similar)

BloodRaine, 2020-08-10:
I am still running into issues when trying to do the same. I am currently using the OpenCVForUnity Package and when I try the same method that @Jntzko tried, I get a sort of namespace or package error - 'Utils' does not contain defenition for (function name here) for multiple methods. I imagine it has something to do with dll's or plugins or something but I'm not certain.

EricVoll, 2020-08-11:
> I am still running into issues when trying to do the same. I am currently using the OpenCVForUnity Package and when I try the same method that @Jntzko tried, I get a sort of namespace or package [...]

I know the error. I guess you used my wrong ros-sharp fork by mistake. 
In the meantime I updated the repository and also added a demo project. I'll edit the readme today with a full tutorial on how to start rosSharp on HoloLens

BloodRaine, 2020-08-11:
Oh thats perfect. My bad on using the wrong one. I got a bit confused with there being 2 in the Repo. Thanks for updating. I'll be sure to update to the new main. Thanks for the assistance!

BloodRaine, 2020-08-12:
Ok, I think it's mostly working. It is now successfully subscribing which is awesome progress! I'm hitting a Nullreference error that an object isnt instantiated, considering everything I have, and the amount of incoming nullReference messages, it is probably something in the subscriber that is trying to be accessed when the message is recieved. Have you run into something like this and have any tips for tracking it down further? Long shot on that one I know, just checking before I go digging for too terribly long.

EricVoll, 2020-08-12:
Yes I experienced that as well but I forgot why.
Generally it's easy to find out that problem. Simply attach Visual Studio to Unity and start the Debugging Mode.
Or in the Unity-Console you'll see the NullReferenceException and double click it to get where the exception is thrown. It probably is obvious then.

BloodRaine, 2020-08-13:
I may be running it improperly as when I run it in debuggin gmode from VS on the device, I dont get any specifics like what object is Null or a line number, and Unity hasn't been giving me any errors when it is run.

EricVoll, 2020-08-13:
So do you get the exceptions if the "game" is running on the holoLens, or while it is running in Editor Mode?
These NullReferenceExceptions usually also occur in Editor Mode.
I had them for some time somewhere in a Subscriber if I remember correctly.

What I did for another project was to do a "sneaky" thing: You can subscribe to the [Unity Log-Output](https://docs.unity3d.com/ScriptReference/Application.LogCallback.html) and add a string publisher that publishes the Log via a ROS-topic. Investing a bit of time upfront in things like this made our life much easier down the road while developing the app.
See the [publisher](https://github.com/EricVoll/ARbotics/blob/master/Unity/Assets/ARRobotInteraction.Base/DebugPublisher.cs) and [here ](https://github.com/EricVoll/ARbotics/commit/398d90ffcf198d8f792accb5e91b790ef4702b5f#diff-146e4ae54435e27b73a931cc9ace547aR20-R26)is how to use it. 

That way you can get Unity's Debug Logs while the app is deployed on the HoloLens.
Theoretically it is possible to attatch the VisualStudio Debugger to a running instance on HoloLens. But I keep forgetting how to do that - you might want to google that for a bit. Its easy once you found out how to...

BloodRaine, 2020-08-13:
So big picture I am using ROS to package up a Pointcloud as an image and send it to unity then render it, we need to do custom stuff with the point cloud so we cant just use the built in spacialization stuff. The app still works fin ine the Editor, and when I run it with "Start Debugging" in VS on the device, so on the hololense, In the bottom left here is a debug window. That was telling me there was a NullReference. But it still ran fine in the Editor. Subscribing to the Unity log might also be a useful thing for debugging for sure.

EricVoll, 2020-08-14:
mhh that seems weird. But its a bit hard to help without the opportunity to look at the stuff myself.
I assume you thought of adjusting the IP-Adresses on the ROS-Connector to whatever your IP-adresses are right?

As I wrote above, try this: https://docs.microsoft.com/en-us/windows/mixed-reality/using-visual-studio#:~:text=Pairing%20your%20device,-The%20first%20time&text=On%20the%20HoloLens%2C%20generate%20a,HoloLens%20to%20dismiss%20the%20dialog.
With the steps explained in this article you can pair your HL with Visual Studio and live-debug with breakpoints etc.

BloodRaine, 2020-08-14:
Yep, I double checked the IP, I'll definately take a look at setting up the HL-VS debug with breakpoints Thanks for reminding me of the link to how to do that. I'll let you know the results

BloodRaine, 2020-08-19:
I was able to get my application working! The issue was that a shader I was using was registerred as Null. In my renderer, I attempt a Shader.find and that was coming back with nothing as I guess it may not be included in the build if nothing references it directly. The solution was to attach the shader to a material in the Resources folder. Thanks for the help. Looks like the data stream is a little slow so my next task will be to try to find a way to speed that up.

EricVoll, 2020-08-20:
Great to hear! So in that case the issue wasn't Ros#.
If you have any further issues with ROS# please create issues over on the UWP fork.