siemens/ros-sharp #283: Introducing ros_csharp_interop

johnwason, 2020-02-19:
Greetings,

I developed a small glue bridge between C# and roscpp for a robot driver project. It uses `topic_tools::ShapeShifter` and SWIG to cross the C++/C# boundary. It currently only supports messages.

I don't plan on developing this any further, but it may be useful if someone wants to continue working on it. It should work with ROS 2.0 but I haven't tested it.

https://github.com/johnwason/ros_csharp_interop

(You will need to build SWIG 4.0 from source on Ubuntu)

johnwason, 2020-02-19:
@LeviArmstrong this may be useful for ROS-I

MartinBischoff, 2020-02-19:
Hi @johnwason 

Please use the issue template when opening new issues in the future.
Is your message a bug report, a question or a feature request?
What is the relation of your message / project to ROS# at all?

Forums like [ROS Discourse](https://discourse.ros.org/) are the right place to advertise projects.
This issue board is for ROS# related issues only.

closed 2020-02-19 by MartinBischoff

johnwason, 2020-02-19:
I want to contribute this functionality to ROS# since I don't want to maintain it. It would give your project the ability to communicate without going through json.

MartinBischoff, 2020-02-20:
Ah, I see your point now.
Thank you for sharing your example and informing community here.

From what I understand of your code after a brief look now is one example to communicate between a C# project and a ROS system without Websocket + JSON (de-)serialization, which drastically reduces latency.

For ROS-Unity applications, this is useful only if Unity runs on the same Ubuntu machine as ROS. In can not be used to communicate with ROS from within .NET applications running on e.g. Windows.

For ROS2-Unity applications this looks like one solid approach. As an alternative to SWIG an topic_tools::ShapeShifter, have you thought about interprocess communication appraoches "to cross the C++/C# boundary"?

johnwason, 2020-02-20:
ROS1 is available on Windows from Micosoft, and I have been using it frequently with the ros_csharp_interop. 

(I think it should be possible to deploy some DLL with the Unity plugin rather than the full ROS installation, assuming the master is running on another machine. If the master is required too, the user will need the full installation.)

http://wiki.ros.org/Installation/Windows

The supplied module also works on Linux using Mono or Core.

The topic_tools::ShapeShifter is used to send and receive binary messages using roscpp. The messages are then serialized/deserialized in C# using BinaryWriter/BinaryReader. The message generation is done using the python script in tools/ for the serialization code. Even with IPC, this would still be necessary. Adding IPC instead of using SWIG is an option, but it would add latency. For a Unity plugin that has a regular update period, you can probably use spinOnce and avoid thread switching entirely. Currently ros_csharp_interop is using a spinner, but that is an easy change to make.