siemens/ros-sharp #284: Joint State Controller that uses collision information using Unity Physics Engine power like simulating in Gazebo

brehman-hiab, 2020-02-20:
* [ x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I followed the wiki and manged to import my robot which is a loader crane (4 DoF robotic arm). 
I manged to move the my robot joints using JoySubscriber in gameObject I called it RosConnect and added JoyAxisJointTransformWriter script within each joint as you shown in the videos. Everything looks the same as I would do it in Rivz. Great! But I would like use Unity Physics Engine power too, like simulating in Gazebo. 

1) I would like to stop moving my robot joints if it collide with it self or environment.
Right now, if my crane collide with ground, it just go through it. But also if I try to move a cube with crane I can kick it around. Means there is collision information but joint state writer doesn't care about it. 
Or 
2) Like in Gazebo, I would like send joint reference to a position controller that can achieve the joint position reference but also taking in account the environment collision. Same as you would do it with joint_state_controller.

I looking forward to your reply.
Bilal
P.S: Really great join with ROS sharp, thanks for amazing work. 

 
---

MartinBischoff, 2020-02-20:
Hi @brehman-hiab ,

your robot model in Unity is fully prepared for a physics simulation. It is based on the same URDF data (rigidbodies, joints, colliders) that Gazebo is using.

Currently, you have an _animation_ where you are visualizing the actual state by explicitly setting the transforms. For a _simulation_ you will have to apply forces to reach the target state.

Please check out the [Unity Simulation Example](https://github.com/siemens/ros-sharp/wiki/User_App_ROS_UnitySimulationExample) in the wiki, where the simulation is performed on the Unity side and the current states are visualized in rviz.

Please check out the [JoyAxisJointMotorWriter](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/MessageHandling/JoyAxisJointMotorWriter.cs) script. With this, you can apply the motor torque of a hinge joint via a Joy axis signal. This can come via a topic from ROS [JoySubscriber](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/JoySubscriber.cs) or via an input axis from Unity [JoyAxisInputPasser](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/MessageHandling/JoyAxisInputPasser.cs).

Based on these scripts you can extend the functionalities that you require for your application.
In the most basic approach you can use the [built-in hinge joint spring-damper functionalities](https://docs.unity3d.com/Manual/class-HingeJoint.html) to reach a target position. For a more physically accurate joint controller model, you will have to implement your own controller and use [AddRelativeTorque](https://docs.unity3d.com/ScriptReference/Rigidbody.AddRelativeTorque.html) around your joint axes.

We didn't add any joint controller scripts in ROS# (yet), as I have the feeling this is very application-specific and it is better to implement / include your own controller code when you are at this point.
If you have a simple and generic joint controller script that you think can be of use for typical ROS# applications please let me know!

brehman-hiab, 2020-02-20:
Thank you for quick and detail reply. I am on it now, i will try to follow steps and get back to you with or without more questions.

brehman-hiab, 2020-03-03:
I worked on my problem, I solved it as I wanted it to be, now I will write how I did it and also another question to you. 
Solution:
Updated unity version I am using 2020.1a25 release, the reason is they have better physics engines especially related to colliders. 

I wrote my own scripts to control the hinge joint anther to control the configurable joint. 
Both scripts take joy axis values from the joy subscribers that you have provided, based on 
[JoyAxisJointMotorWriter.cs](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/MessageHandling/JoyAxisJointMotorWriter.cs), the only difference for hinge joint, I used spring, where set spring and dumping to keep the joint its position and changed the target value to an angle that I wanted to move to. 

For prismatic movement, I used a Configurable joint. For the Configurable joint, I locked the motion and rotation in all directions except the x Motion set to free. 
Then I changed Position Spring, Position Damper and Maximum Force in a way that I can achieve linear motion by setting the target position, like move 1 meter in Target Position.X.  In all these two scripts, I am mainly depended on the joy subscribers, which is subscribed to a ROS topic. 

Finally, now I can move my robot each joint as I want in Unity Editor, and play my simulation game super happy :D 

Now Question: 
Once I build the Unity to create a standalone game application, I can see in Linux that a client is connected to ROS WebSocket, great, but it seems like the joy subscriber is not getting the ROS topic updated values. Just to double-check, I use inputs from Edit > Project Settings > Input, and I can move my robot with my keyboard keys but not with joystick topic that ROS is publishing :( 
I did some research on unity input and also look into your wiki like you explained in 
[Animate a Robot Model in Unity](https://github.com/siemens/ros-sharp/wiki/User_App_NoROS_AnimateRobotModelInUnity3D). but no luck so far. Is there anything I might be missing? or do I need to create custom joystick in Unity and write my joy topic values to it? Looking forward to your reply. 
best regards
Bilal

closed 2020-03-10 by brehman-hiab