siemens/ros-sharp #287: Android ROS connection

gapostolopooulos, 2020-02-25:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I am trying to connect an Android device (tablet) to ROS using Ros#. I imported rossharp to my unity project. . Althought the Unity-Ros connection is fine and I can publish my topics on Ros when I build my project and run it on the tablet it connects  with ROS but the topics don't publish  to ROS. Do you have any idea  what might be  the problem?

---
* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is:

MartinBischoff, 2020-02-26:
Hi @gapostolopooulos ,

I didn't try ROS# for Android myself and am currently not planning to, so likely there are others who can give you more profund answers than mine: Most likely it's because of the Newtonsoft.JSON dependency which is probably not compatible with Android systems. Please check out https://github.com/siemens/ros-sharp/issues/278 which suggests replacing it by [JSON.NET for Unity](https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347). This dependency is also compatible with Android.
I hope this helps.
Android application experts, please revise my suggestion!

gapostolopooulos, 2020-02-27:
Hi @MartinBischoff ,
Thank you for your immediate answer. Unfortunately I have already tried this and doesn't work. I still have the same problem.

odydoum, 2020-02-27:
Hello, do logs contain anything of interest? I am not familiar with Android, but I found [this](https://forum.unity.com/threads/how-to-capturing-device-logs-on-android.528680/).

berkayalpcakal, 2020-03-26:
Hello @gapostolopooulos ,
With the [recent changes](https://github.com/siemens/ros-sharp/pull/292) in the repo, it is possible to use .NET API for JSON serialization. I wonder if it resolves your issue with Android-ROS communication. Could you please try the latest version of ROS# in your case and share your experiences here?

joelmaca96, 2020-04-09:
Hello @gapostolopooulos. I had the same issue with unity 2019.3.7 and this asset solved my problems:
https://assetstore.unity.com/packages/tools/input-management/json-net-converters-simple-compatible-solution-58621

I don´t know if it works at 100% but at least the ros publisher in the app apears in the rostopic list in the server side. Hope that it helps

berkayalpcakal, 2020-04-09:
I have got some concrete news!

After having the .NET Json serialization tool in ROS# with #292, I managed to send and receive simple string messages between ROS and Android. I simply set the `Serializer-Protocol` settings to `Microsoft - WebSocketSharp` on the inspector window of `RosConnector`. 

I am planning to test the communication with more complex messages like compressed images, joint states, odometry etc. I will comment my further experiences here.

berkayalpcakal, 2020-04-12:
With the settings I mentioned above I successfully subscribed to the joint states from ROS to Android, and publish compressed images from Android to ROS. 

[image]

I think the reason for the issue at the first place was that Newtonsoft Json serialization does not work with Android devices. You should use .NET Json serialization tool (see #292) with `WebSocketSharp` protocol instead.

As @odydoum  suggested [here](https://github.com/siemens/ros-sharp/issues/287#issuecomment-592137725) already, [this link ](https://forum.unity.com/threads/how-to-capturing-device-logs-on-android.528680) helps. Thanks a lot!

closed 2020-04-12 by berkayalpcakal

luisfusim, 2020-10-16:
> this link

Sorry but just to confirm, with this technique is posible juts build ARM7 because MONO doesn't support ARM64 and IL2CPP is no compatible.
Thanks