siemens/ros-sharp #288: URDF model does not appear in Unity 2019.2x

pegasus0642, 2020-02-26:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10
* My Unity Version is: 2019.2.2
* My ROS Distribution is: Kinetic
* My Build Target Platform is: Unity 2019.2.2

**Here is my bug description:**
I followed the instructions and modified the publish_description file for UR10 robot arm. Everything is fine but I could not see the model in Unity. There is no error or fault during the transferring.
[image]
[image]
[image]
[image]

Perform the following steps reproduce the bug:
1- roslaunch file_server publish_description_ur10.launch
2- Read robot description from Unity

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```

MartinBischoff, 2020-02-28:
Hi @pegasus0642 , we could not reproduce the bug you reported:
[image]

Everything looks fine on the ROS side. We can isolate the error you have to the URDF import on the Unity side. How does the rest of your ur10 GameObject look like? Is only the visualization missing? Are there meshes in the MeshFilters? Are there colliders? Do you use any other assets beside ROS# that might have an impact on the the dae import?

pegasus0642, 2020-03-02:
Hi Martin,

Thanks for the reply. I did several tests with Unity and you are right, ROS# does not have a bug. When I try to combine MRTK and ROS#, the robot model does not appear. However, in the fresh project, firstly I need to transfer the URDF models into unity and then I need to add MRTK or additional plug-ins. In this way, it is working as it should be.   Your comment was helpful for me, now everything is working fine.

Thanks a lot.

closed 2020-03-02 by pegasus0642

MartinBischoff, 2020-03-02:
Hi @pegasus0642 ,

does Mixed Reality Toolkit (MRTK) also implement an [AssetPostprocessor](https://docs.unity3d.com/ScriptReference/AssetPostprocessor.html) like ROS# is doing e.g. [here](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/Urdf/Editor/MeshProcessing/ColladaAssetPostProcessor.cs)? This might cause the incompatibility.