siemens/ros-sharp #289: Enable CBOR decompression

Changliu52, 2020-02-27:
## I have a feature request! ##

The rosbridge has just developed a really nice feature to integrate CBOR compression for pointcloud. https://github.com/RobotWebTools/rosbridge_suite/pull/452
But at the moment ros-sharp cannot receive it. I think it will be really valuable to enable decompression in ros-sharp to utilise this feature. https://github.com/RobotWebTools/rosbridge_suite/issues/453

How would you think this is can be enabled or adapt to work towards streaming pointcloud2 through CBOR compression? 

Best Regards,
Chang

berkayalpcakal, 2020-04-01:
Hey @Changliu52 , 

I am trying to deserialize CBOR encoded messages as well. Do you have any progress so far? I am having an issue which I explain [here ](https://github.com/dahomey-technologies/Dahomey.Cbor/issues/56)in detail, which is basically I could only get corrupted json strings when I decode cbor-compressed PointCloud2 messages so I cannot deserialize the messages into custom C# objects.

Changliu52, 2020-04-11:
Sorry, no progress update so far.

MartinBischoff, 2024-01-22:
We restrict ROS# serializer support to Microsoft.JSON and Newtonsoft.JSON only.
There is the generic ISerializer interface with which application-specific serializers can be implemented.

closed 2024-01-22 by MartinBischoff