siemens/ros-sharp #291: Image Subscriber produces a red "?"

Cpschorr, 2020-03-09:
My goal is to use the image subscriber to take data from a turtlebot2's camera and display it on an object in unity. I've subscribed to topic camera/rgb/image_raw and placed the mesh of the object I want to update and a sprite which is tied to the material of the object in the two fields Mesh Render and Texture 2D. 
I'm also confused on line 33.
meshRenderer.material = new Material(Shader.Find("rosscreenmat.mat"));
I tried both leaving it as is or putting in the name of the material that was attached to the object (rosscreenmat.mat)

Looking at my topics in Rviz, I can see the images are coming through and I've gotten the odometry subscriber to work, so I believe the issue is just with the image subscriber script or how I'm using it. Thank you!

berkayalpcakal, 2020-03-25:
Hello @Cpschorr,

Could you please clearly write down the steps to reproduce the issue? When you open an issue, you should see an issue template, and there is a reason for that.

At first glance, if you are using the [ImageSubscriber.cs](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/ImageSubscriber.cs), you should subscribe to the topic **"camera/rgb/image_raw/compressed"** (please double check, it should be a compressed image), since the image subscriber mentioned above subscribes to a topic of CompressedImage message type. Please take a look at the [gazebo simulation scene](https://github.com/siemens/ros-sharp/tree/master/Unity3D/Assets/RosSharp/Scenes), there should already be an example regarding the use case of ImageSubscriber. 

If you have further issues, please open an issue using the issue template.

closed 2020-03-25 by berkayalpcakal

MarcoDotIO, 2021-12-08:
Hello,

I am having a similar issue on my end; since I am using a Tello Drone. I made sure to subscribe to the /tello/image_raw/h264 topic; and had made sure the data is paritied on both ends. Here are my steps to reproduce the issue:

- Make new Unity project with ROS Sharp
- Make a new 2D UI Image (And added a mesh renderer component)
- Make a new script asset with the contents of ImageSubscriber.cs
- Made sure the Image object was connected to the Mesh Renderer variable
- Played the scene
- Inside the mesh for the image, there will be a red question mark set as the albedofor the material.

I'm using Unity 2020.3.23f1, latest version of ROS Sharp, and ROS Melodic.

berkayalpcakal, 2021-12-08:
Hello @MarcoDotIO,

It seems like you are indeed having the similar problem; you are using [ImageSubscriber](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/ImageSubscriber.cs) for subscribing to the message type of [sensor_msgs/Image](http://docs.ros.org/en/api/sensor_msgs/html/msg/Image.html) where you should aim for subscribing to the type of [sensor_msgs/CompressedImage](http://docs.ros.org/en/api/sensor_msgs/html/msg/CompressedImage.html). I believe the reason for red question mark is that the received data cannot be decoded properly.

I suggest that you first try converting raw images to compressed images via [image_transport](http://wiki.ros.org/image_transport) within ROS before sending over to Unity. 

I hope this helps. Maybe close the other issue #414, because this issue page already covers the similar problem?

MarcoDotIO, 2021-12-08:
> Hello @MarcoDotIO,
>
> It seems like you are indeed having the similar problem; you are using [...]

Hello there,

Thank you for the reply. I had made sure that the topic is using the sensor_msg/CompressedImage topic. Here is the ROSWiki page for the driver I'm using for the [Tello drone](http://wiki.ros.org/tello_driver). It does appear that is isn't a compressed image topic, as the page says it's a [H264_image_transport/H264Packet](https://github.com/tilk/h264_image_transport/blob/master/msg/H264Packet.msg) topic. Though when running it on hardware, running `rostopic info /tello/image_raw/h264` reveals that it is a sensor_msgs/CompressedImage topic. These seem to be contradictory, as shown with Unity being unable to process the image. If you can help out, let me know.

samMallick, 2022-01-14:
I had this problem too. To fix it I used a republishing node on the ROS end to change the transport method from 'h264' to compressed'. 

cmd: roslaunch image_transport republish h264 in:=/tello/image_raw compressed out:=/newTopic

Then by subscribing to /newTopic/compressed in Unity you can process the image using the normal ImageSubscriber