siemens/ros-sharp #296: ROS# does not work when building the project

Scharfhausen, 2020-03-22:
Hi there,

I've been trying to communicate from Unity to ROS in several ways:

1. Clicking play button from the Unity3D app --> The communication works, it connects to rosbridge server and the desired topic can be displayed
2. Building the project without changing anything from 1) --> I tried to export it for Windows (.exe) as well as for Linux (.x86-64), but none of them worked. Rosbridge server notices there is one new connection, but "rostopic list" does not show the Unity topic.

Steps Taken: 

I have tried to replace the Newtonsoft.xml from this page: https://assetstore.unity.com/packages/tools/input-management/json-net-for-unity-11347, which does not work.

I don't understand what's the difference between compiling+building the project and running it from the app.

Andor233, 2020-03-23:
You should change your project to mono & .net 4.x 
But if someone replaces  Newtonsoft.Json to JSON .NET it will helpful.
Newtonsoft.Json is bad for cross compile and li2cpp support

Scharfhausen, 2020-03-23:
> You should change your project to mono & .net 4.x
> But if someone replaces Newtonsoft.Json to JSON .NET it will helpful.
> Newtonsoft.Json is bad for cross compile and li2cpp support

Thank you very much, Andor233, I would've never thought of that, but is finally working. Have a great day!

Andor233, 2020-03-24:
I'm happy to hear you solved your problem, it takes me much time when I faced the problem, but it just a temporary solution.
Newtonsoft.Json can't work on the platform which doesn't support reflection.
So maybe keep this issue open maybe better?

berkayalpcakal, 2020-03-24:
Hi,
With the recent changes [here](https://github.com/siemens/ros-sharp/pull/292), one can use .NET JSON serializer instead of Newtonsoft JSON serializer. Maybe you can give it a try and see if it works on the platform you want. 
It would be nice to see your experiences regarding which serializer works with which platform, so you can comment your findings on this issue.

closed 2020-03-24 by berkayalpcakal

Scharfhausen, 2020-03-24:
Hi there, berkayalpcakal

I am not sure I'm doing this correctly, so could you please tell me how to use .NET JSON instead?

Scharfhausen, 2020-03-24:
> Hi,
> With the recent changes [here](https://github.com/siemens/ros-sharp/pull/292), one can use .NET JSON serializer instead of Newtonsoft JSON serializer. Maybe you can give it a try and see if it [...]

Hi,

I am not sure I'm doing it correctly. Could you please guide me through the process. I think I am not replacing anything, because results are the same: it does not work once built.

berkayalpcakal, 2020-03-24:
Hi @Scharfhausen,

Have you pulled the recent changes to your local already?
If you did so, you can see that [RosSocket](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/RosSocket.cs#L39) gets an argument to set the serialization tool. 

What you can do on Unity side;
- Go to your RosConnector component on the inspector editor
- Set e.g. Serializer: Microsoft and Protocol: WebSocketSharp etc.

Scharfhausen, 2020-03-24:
> Hi @Scharfhausen,
>
> Have you pulled the recent changes to your local already?
> If you did so, you can see that [...]

OK, I did not know that it was **that** recent, I cloned the repo 10 days ago. But with the new commit, it works perfectly selecting "Microsoft" as serializer, both in Linux (I tried with Ubuntu 16.04) and in Windows 10.