siemens/ros-sharp #297: Service server fails to return result on Ubuntu

jameseclarke, 2020-03-24:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Ubuntu 18.04
* My Unity Version is: 2019.3.5f1
* My ROS Distribution is: Melodic
* My Build Target Platform is: Linux x86_64

**Here is my bug description:**
I've built a simple add_two_ints service client in ROS with the source code copied from the ROS tutorials, and tested that it works properly. The problem comes when trying to write the service server in Unity. The server is able to advertise the service correctly using the RosBridge protocol, and it does receive the service call, however the client never gets the result when the ServiceCallHandler function exits.

Perform the following steps reproduce the bug:
1. Build the catkin workspace provided in the attached zip file (or alternatively complete the roscpp service tutorial.
2. Generate RosSharpMessages for the beginner_tutorials package in the workspace.
4. Source the catkin workspace and launch the rosbridge_server rosbridge_websocket.launch file.
3. Add the below code into a Unity script, add it to an empty GameObject, name the service, and run the game.
4. Source the catkin workspace in another terminal and call the service using rosservice.

Observed results:
The Unity debug log shows the request arriving with the correct arguments, and the response value being set correctly. However, the terminal from which the service is called never receives the result. This occurred on both Ubuntu 16.04 and 18.04, with ROS Kinetic and Melodic respectively.

Expected results:
The service should return a result.

Relevant Code:
```
﻿using System.Collections;
using System.Collections.Generic;
using UnityEngine;

namespace RosSharp.RosBridgeClient
{
	public class AddTwoIntsServiceProvider : UnityServiceProvider<MessageTypes.BeginnerTutorials.AddTwoIntsRequest, MessageTypes.BeginnerTutorials.AddTwoIntsResponse>
	{
		protected override bool ServiceCallHandler(MessageTypes.BeginnerTutorials.AddTwoIntsRequest request, out MessageTypes.BeginnerTutorials.AddTwoIntsResponse response)
		{
			response = new MessageTypes.BeginnerTutorials.AddTwoIntsResponse(request.a + request.b);

			Debug.Log("Service called with a=" + request.a.ToString() + " b=" + request.b.ToString());
			Debug.Log("Returning sum=" + response.sum.ToString());

			return true;
		}
	}
}
```
[addtwoints_ws.zip](https://github.com/siemens/ros-sharp/files/4374593/addtwoints_ws.zip)

Rosthouse, 2020-03-26:
I'm facing exactly the same issue right now. Furthermore, I'm able to call Services implemented in rspy FROM Unity and receive the correct results.

I have also tested if it's a general problem with rosbridge_server and implemented a simple topic in Unity:

```csharp
using System;
using System.Collections;
using RosSharp.RosBridgeClient;
using RosSharp.RosBridgeClient.MessageTypes.RosPythonTopic;
using UnityEngine;

[RequireComponent(typeof(RosConnector))]
public class AgePublisher : MonoBehaviour
{
    [SerializeField] protected string topic;
    protected RosConnector connector;
    protected string connectionId;
    private int counter = 0;
    private string name = "Mark";

    public void Start()
    {
        this.connector = GetComponent<RosConnector>();
        this.Advertise();
    }
    public void Advertise()
    {
        if (!this.connector.IsConnected.WaitOne(5000))
        {
            Debug.Log("Connection to ROS failed");
            return;
        }
        this.connectionId = this.connector.RosSocket.Advertise<Person>(this.topic);
        Debug.Log($"Advertised {this.topic}[{this.connectionId}]");
        this.StartCoroutine(this.PublishPerson());
    }

    public void Unadvertise()
    {
        if (!String.IsNullOrEmpty(this.connectionId))
        {
            this.connector.RosSocket.Unadvertise(this.connectionId);
            this.connectionId = null;
            this.StopCoroutine(this.PublishPerson());
            Debug.Log($"Unadvertised {this.topic}[{this.connectionId}]");
        }
    }

    public IEnumerator PublishPerson()
    {
        while (true)
        {
            this.connector.RosSocket.Publish(this.connectionId, new Person(this.name, this.counter));
            this.counter++;
            yield return new WaitForSeconds(1);
        }
    }
}
```

And Person is a simple `msg` generated from this:

```
string name
int32 age
```

My findings therefore are that the general communication over ROS does work (as I can publish messages from Unity into ROS and receive them there:

```bash
$ rostopic echo /mark_publisher
name: "Mark"
age: 69
---
name: "Mark"
age: 70
---
name: "Mark"
age: 71
---
name: "Mark"
age: 72
---
...
```

closed 2020-03-27 by berkayalpcakal

berkayalpcakal, 2020-03-27:
Hello,

The problem is the "id" field of [**Call Service**](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#346-call-service) does not match with the "id" field of [**Service Response**](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#349-service-response), which results in basically ignoring the response.

#298 resolves this issue. Please pull the recent changes in the repo to your local, and do not forget to update the RosBridgeClient.dll in your Unity3D project.

Thanks for identifying the bug.

Rosthouse, 2020-03-27:
And thanks for the quick fix. Was just able to confirm that it solved the problem on my end. Huge thanks :)

commit 356319d9 (2020-03-27): bug fix #297, update dll

"id" filed in "Call Service" was not mathcing with the "id" field in the "Service Response"
commit 4ef962b3 (2020-03-27): Merge pull request #298 from berkayalpcakal/ServiceResponseBug

bug fix #297, update dll