siemens/ros-sharp #300: Versions not match when I import the lastest dll

Andor233, 2020-04-07:
* My OS is: Win10 & manjaro
* My Unity Version is: 2020.1
* My ROS Distribution is: melodic
* My Build Target Platform is: Linux

When I try to import the lastest dll `commit 4ef962b` ros-sharp, unity can't restore the package, it says:
```
projectAssembly 'Assets/RosSharp/Plugins/RosBridgeClient.dll' will not be loaded due to errors:
RosBridgeClient references strong named System.Threading.Tasks.Extensions, versions has to match. Assembly references: 4.2.0.1 Found in project: 4.2.0.0.
```
Almost all the dll under Microsoft has same problem, I try to compile by myself, but not work.

berkayalpcakal, 2020-04-07:
Hello @Andor233,

The commit [4ef962b](https://github.com/siemens/ros-sharp/commit/4ef962b32c675dfc416f96cbe8e11d9d2fb6411e) only changes the `RosSocket.cs` and updates the corresponging dll `RosbridgeClient.dll`. There should not be any changes in the versions of any other plugins. Are you sure that you refer to correct commit?

Are you trying to build the Unity project solution or the ROS# library project solution? Could you please provide the full error logs, and steps to reproduce the issue?

Andor233, 2020-04-08:
Sorry, I mean I'm using f3f6b542392ed7d44342f2070c3c2890b0c5944c before, and when I update to the latest because I want to try the Microsoft JSON, it happens. 

It is not building problem, I can't import ros sharp when I open editor, here is the log
```

Begin MonoManager ReloadAssembly
Assembly 'Assets/RosSharp/Plugins/RosBridgeClient.dll' will not be loaded due to errors:
Reference has errors 'System.Text.Json'.
Assembly 'Assets/RosSharp/Plugins/Microsoft/Microsoft.Bcl.AsyncInterfaces.dll' will not be loaded due to errors:
Microsoft.Bcl.AsyncInterfaces references strong named System.Threading.Tasks.Extensions, versions has to match. Assembly references: 4.2.0.0 Found in project: 4.2.0.1.
Assembly 'Assets/RosSharp/Plugins/Microsoft/System.Memory.dll' will not be loaded due to errors:
System.Memory references strong named System.Runtime.CompilerServices.Unsafe, versions has to match. Assembly references: 4.0.4.1 Found in project: 4.0.6.0.
System.Memory references strong named System.Buffers, versions has to match. Assembly references: 4.0.2.0 Found in project: 4.0.3.0.
Assembly 'Assets/RosSharp/Plugins/Microsoft/System.Text.Encodings.Web.dll' will not be loaded due to errors:
Reference has errors 'System.Memory'.
Assembly 'Assets/RosSharp/Plugins/Microsoft/System.Text.Json.dll' will not be loaded due to errors:
System.Text.Json references strong named System.Threading.Tasks.Extensions, versions has to match. Assembly references: 4.2.0.0 Found in project: 4.2.0.1.
System.Text.Json references strong named System.Buffers, versions has to match. Assembly references: 4.0.2.0 Found in project: 4.0.3.0.
Assembly 'Assets/RosSharp/Plugins/Microsoft/System.Threading.Channels.dll' will not be loaded due to errors:
System.Threading.Channels references strong named System.Threading.Tasks.Extensions, versions has to match. Assembly references: 4.2.0.0 Found in project: 4.2.0.1.
Assembly 'Assets/RosSharp/Plugins/Microsoft/System.Threading.Tasks.Extensions.dll' will not be loaded due to errors:
System.Threading.Tasks.Extensions references strong named System.Runtime.CompilerServices.Unsafe, versions has to match. Assembly references: 4.0.4.1 Found in project: 4.0.6.0.
```
Something I miss?

berkayalpcakal, 2020-04-09:
Hi again @Andor233,

I was able reproduce the issue only with Unity version 2020.1. It seems that there are conflicts in assembly versions of the plugins. I did some effort to resolve the issue but I was not successful. Here is what I have done:

* I removed all the plugins under the folder `ros-sharp/Unity3D/Assets/RosSharp/Plugins/Microsoft` and installed those packages via NuGet package manager in Unity3D Visual Studio solution, also updated already installed packages. 
* I opened the Visual Studio solution for the library under the folder `ros-sharp/Libraries` and again updated packages via NuGet package manager.  
* I built the library solution in release mode, and replaced the `RosBridgeClient.dll` with the ones in the 'Unity3D/Assets/RosSharp/Plugins'

However I am still getting the same error.

Have you tried using the latest official release of Unity (2019.3) instead of pre-release (2020.1)? I suggest not to use the beta version of Unity for the time being. I would be glad if you could comment your further experiences with the issue here.

Andor233, 2020-04-10:
Yes, I only try the 2020.1, thankyou your try, if I find a way to fix it, I will let you know.

Andor233, 2020-04-14:
@berkayalpcakal 
It seems like unity has built-in this dll but with different version after 2020.1. 
Do you have any good solution for it? Maybe build it with the same version as unity use?

Andor233, 2020-04-14:
@berkayalpcakal I found the solution, thank your help.
 I think it's unity's mistake. It seems like unity adds a function validate references when you import dll. Which maybe works incorrectly, after I uncheck it, everything works fine.
So how could I tell the Unity their mistake? Can you help me?
> PS: It seems like unity add the `unsafe` dll in their collection packages, so it will be conflicted

closed 2020-04-14 by Andor233

berkayalpcakal, 2020-04-14:
Hey @Andor233,

I am glad that you resolved the issue. Could you please clearly write down the steps you took here? It might be a nice reference for others.

> By the way, what are the glasses you wear in your avatar?

It is an AR glass from Meta Vision. 

> So how could I tell the Unity their mistake? Can you help me?

I am not sure, maybe [this](https://support.unity3d.com/hc/en-us/articles/206336985-How-do-I-submit-a-bug-report-) helps.

Andor233, 2020-04-15:
After importing the ros-sharp, get into the `Plugins` folder, and close all your dlls' `validate references` like this.
[image]

Your glasses are very cool.

jespersmith, 2020-04-28:
@Andor233 Thank you for the tip.

@berkayalpcakal Would it make sense to make a PR with "validate references" turned off?

Andor233, 2020-04-28:
@jespersmith 
 I can do it if you want. It’s easy, just change the .meta

jespersmith, 2020-04-29:
That would be great. 

It's easy to fix, but not very obvious when importing. I was going to file a bug report but saw your bug report first.