siemens/ros-sharp #301: URDF is not loaded after connecting to ROS

Rosthouse, 2020-04-07:
* [x ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---

## I found a bug! ##
* [ x] I am using the latest ROS# version available here on the master branch.
* [x ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10
* My Unity Version is: 2019.3.3f1
* My ROS Distribution is: Melodic
* My Build Target Platform is: Ubuntu 18.04

**Here is my bug description:**

When trying to load a URDF file from ROS to Unity, RosBridgeClient only manages to connect to ROS, but doesn't read any data from the parameter server.

**Perform the following steps reproduce the bug:**

1. Create a package containing the launch file and the urdf file (in the launch and urdf folder respectively). These are listed below. 

> __*Note:*__ For this description, the package is called `moving_cube` and the launch file is called `publish_moving_cube_urdf.launch`.

2. Build your catkin workspace (`catkin_make`)
3. Source your terminal (`source devel/setup.bash`)
4. Use `roslaunch` to start the previously created package:

```
$ roslaunch moving_cube publish_moving_cube_urdf.launch
```

5. The summary of your start should read as follows:
```
SUMMARY
========

PARAMETERS
 * /robot/name: moving_cube
 * /robot_description: <?xml version="1....
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address: 0.0.0.0
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosbridge_websocket/unregister_timeout: 10
 * /rosbridge_websocket/use_compression: False
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    file_server (file_server/file_server)
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)
```

6. In Unity, open the Window **URDF Transfer (From ROS tun Unity)**
7. Verify your connection settings
8. Click "Read Robot Description"

Observed results:

Only the **Connected** field reports that it is done. All subsequent process steps remain `open`:

[image]

The console lists the following error: 

```
Not all resource files have been received before timeout.
UnityEngine.Debug:LogWarning(Object)
RosSharp.RosBridgeClient.TransferFromRosHandler:ImportAssets() (at F:/<redacted>/ros-sharp/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/Editor/UrdfTransfer/TransferFromRosHandler.cs:93)
RosSharp.RosBridgeClient.TransferFromRosHandler:TransferUrdf(Protocol, String, Int32, String, String, SerializerEnum) (at F:/<redacted>/ros-sharp/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/Editor/UrdfTransfer/TransferFromRosHandler.cs:70)
RosSharp.RosBridgeClient.TransferFromRosEditorWindow:<OnGUI>b__9_0() (at F:/<redacted/ros-sharp/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/Editor/UrdfTransfer/TransferFromRosEditorWindow.cs:100)
System.Threading.ThreadHelper:ThreadStart()
```

Expected results:

A new GameObject should have been created. 

Relevant Code:

I created a launch file, that puts the URDF into the parameter server of ROS. I verified that theses parameters are available both by using `rosparam ...` on the (ROS) host machine and by writing a simple `node.js` script that gets the values from these parameters on my client machine (where Unity is also running).

The launch script is as follows:

```
<?xml version="1.0" encoding="UTF-8"?>
<launch>
 	<include file="$(find file_server)/launch/ros_sharp_communication.launch">
		<arg name="port" value="9090" />
	</include>   

	<arg name="x" default="0.0" />
	<arg name="y" default="0.0" />
	<arg name="z" default="0.11" />
	<arg name="roll" default="0"/>
	<arg name="pitch" default="0"/>
	<arg name="yaw" default="0.0" />
	<arg name="urdf_file" default="$(find moving_cube)/urdf/moving_cube.urdf" />
	<param name="robot/name" value="moving_cube" />
	<param name="robot_description" textfile="$(arg urdf_file)" />
</launch>
```

And the URDF is as follows:

```
<?xml version="1.0" encoding="UTF-8"?>
<robot name="moving_cube">
	<link name="cube_body">
		<inertial>
			<origin xyz="0 0 0" rpy="0 0 0"/>
			<mass value="0.5" />
			<inertia ixx="0.00333333333333" ixy="0.0" ixz="0.0" iyy="0.00333333333333" iyz="0.0" izz="0.00333333333333"/>
		</inertial>
		<collision>
			<origin xyz="0 0 0" rpy="0 0 0"/>
			<geometry>
				<box size="0.2 0.2 0.2"/>
			</geometry>
		</collision>
		<visual>
			<geometry>
				<box size="0.2 0.2 0.2"/>
			</geometry>
		</visual>
	</link>
</robot>
```

berkayalpcakal, 2020-04-08:
Hello @Rosthouse,

I was able to reproduce the issue. It seems that there is a problem when using `WebSocketNet` protocol with `Microsoft` serializer. Could you please do the following and see if it works;

* Launch your file on ROS side as you already have done
* In Unity, open the Window URDF Transfer (From ROS tun Unity)
* Click on the button `Reset to Default`
* Adjust IP address.
* Click on `Settings` field.
* Change `protocol` - `serializer` to either **WebSocketSharp - Microsoft** or **WebSocketSharp - Newtonsoft**
* Click on `Read Robot Description`

The robot should now be transferred to Unity successfully. If this resolves your problem please close the issue.

Rosthouse, 2020-04-08:
Can confirm this worked, using `protocol: WebSocket Sharp`. Tested both with `serializer`s, worked with both.

Thanks for the answer :)

closed 2020-04-08 by Rosthouse