siemens/ros-sharp #302: Wheels bend when turning

stevemartinov, 2020-04-20:
[*] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
[*] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

 [*] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I have imported my robot with mecanum wheels with 9 rollers on each wheel. Kinematics more or less looks correct to me but for some reason, my wheels are not rigid enough (or the joints) as seen on the video here: [LINK](https://drive.google.com/file/d/1FBycKDUAW9agLjiVda880WS7k0w8oOug/view?usp=sharing)
The wheels start to bend like there is not enough rigidness. Is it really possible that the PhysX engine took the location of the wheels into consideration?

---

MartinBischoff, 2020-04-20:
Hi @stevemartinov  

> Is it really possible that the PhysX engine took the location of the wheels into consideration?

Sorry, I do not understand this question. Also please ask Unity/PhysX problems in the forums linked up in the issue template. You will get better help there.

Nevertheless, Here are some hints that I would suggest after watching the video:
* In Time Settings: Reduce fixed time step
* In Physics Settings: increase default solver iterations and default solver iterations.
* Check the colliders, CoMs, masses and inertias of your model.

Best,
Martin

stevemartinov, 2020-04-21:
I found my problem. I had a "sphere" collision type for my wheels which means they could collide and move in all direction.

closed 2020-04-21 by stevemartinov