siemens/ros-sharp #303: How to publish PointCloud from Unity to ROS?

stevemartinov, 2020-04-22:
* [* ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [* ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [* ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I want to publish the pointcloud from the Unity to ROS and I almost have everything for that. So as I attach the camera, I can get Depth and RGB (correct?).

This package already can send `sensor_msgs::PointCloud2` but the problem is that the data has to be in bytes[] format. I also saw the script from this repo and this for loop seems to be doing a correct thing:

`
 Points[u + v * width].z = depth * invFocal;
                        Points[u + v * width].x = u * depth * invFocal;
                        Points[u + v * width].y = v * depth * invFocal;
                        Points[u + v * width].rgb = new int[] { 0, 0, 0 };// rgbImage[u, v];
`

That is much better as we can divide data in X, Y and Z but with bytes that is one hell of a task for me. How can it be done correctly?

stevemartinov, 2020-04-24:
I haven't yet managed to send any data to ROS. So if I have an depth image from the unity camera of 640 x 480, it means I need a point cloud data array of size of => 640 * 480 * 6 (because X,Y,Z,R,G,B). Then divide that size by 6 and first populate X values up to SIZE / 6 and etc?

stevemartinov, 2020-04-24:
Also, I just realized that the format is actually byte[]. So correct me if I am wrong but the actual PC size will be 640 * 480 * 6 * 8

berkayalpcakal, 2020-04-24:
I think this is wrong. You need floats to store X, Y, Z values. 
You need to have 3 floats + 3 uint8, so in total 15 bytes are needed for a single point in this case.
That is why you have to use BitConverters to convert 4 bytes of a float object to array of 4 bytes, so that you can place them in `data` field in the pointcloud2 message.

[Here](https://github.com/siemens/ros-sharp/issues/152#issuecomment-446616950) is an example of an interpretation of `data` field for getting a meaningful XYZRGB values.

MartinBischoff, 2020-05-22:
considering this issue resolved by @berkayalpcakal .

closed 2020-05-22 by MartinBischoff