siemens/ros-sharp #304: Vive Ros connection problem

Reim6118, 2020-04-23:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I'm using Ros# in my unity project trying to connect HTC Vive Focus with Ros.

It worked all fine when I connect Unity with Ros but when i build it as an app and put it into Vive, it won't publish to the Ros topic . I found it's similar to #287 and #235 so I tried the settings mentioned there by changing the **serializer-protocol** to **Microsoft-Websocket sharp** but the result remains.

The debug log shows: **NullReferenceException: Object reference not set to an instance of an object  at RosSharp.RosBridgeClient.TwistedPublisher.UpdateMessage () [0x00000] in <00000000000000000000000000000000>:0** 

Is there anything I missed or doing wrong

---
My OS is: Windows 10

My Unity Version is: 2019.3.6f1

My ROS Distribution is: Kinectic

My Build Target Platform is: HTC Vive Focus

berkayalpcakal, 2020-04-23:
Hello @Reim6118,

Unfortunately I do not have HTC Vive Focus so I cannot reproduce this issue by myself. According to the error log that you mentioned, probably the error is raised when `Publish(msg)` is called inside the publisher script. If we have a look at the [corresponding script](https://github.com/siemens/ros-sharp/blob/a3238ced8ed533caeda1062625979cabb4ad78b7/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnityPublisher.cs#L36), I can guess that either `rosConnector` is null or ,with a higher chance, `publicationId` is null due to missing advertisement.

I can only suggest that you include the entire source code of RosBridgeClient into Unity3D/Assets, and run the application in debug mode to see at which point the error occurs.

FraDePace, 2020-07-02:
I have a similar issue. I'm using:

- HTC Vive
- Unity 2019.1.9

When I build the application, I cannot subscribe to /joint_states topic using JointStateSubscriber.
I checked the UnitySubsciber and the rosConnector is connected but it fails during the subscription. The method rosConnector.RosSocket.Subscribe<T> returns an empty string (or it keeps stuck, not sure) instead of returning the topic name.

In Editor mode, everything works fine.

rassel, 2020-10-22:
I have a similar issue too.
-> [Play (Preview mode )] can subscribe topic, but [Build And Run] mode cannot subscribe topic.

I found a workaround and wrote it at #346 
Please refer to it.

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff