siemens/ros-sharp #305: Issues with transferring URDF to Unity while following wiki tutorial

ChrisMOxon, 2020-04-24:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: <Ubuntu 16.04.6>
* My Unity Version is: <2018.4.18>
* My ROS Distribution is: <Kinetic>
* My Build Target Platform is: <not decided yet>

**Here is my bug description:**
I've been following the tutorial in the wiki and run into several issues trying to import the URDF to ROS

Perform the following steps reproduce the bug:
Follow all steps of wiki tutorial up to https://github.com/siemens/ros-sharp/wiki/User_App_ROS_TransferURDFFromROS

Observed results:
Firstly you run into an error while running the roslaunch file_server regarding tornado not being installed, this is solved via 
python -m pip install tornado
Installing tornado (has to be below version 6) allows the roslaunch command to run without issues.  

In Unity now when you open RosBridgeClient, select "Transfer URDF from ROS", enter the correct IP address and click "Read Robot Description" you can see it connects.  The text in Unity next to Connected: goes from open to done and in the terminal on the ROS machine you see "Client Connected. 1 clients total".  All of the other steps in Unity remain red "open" and no URDF is imported.

I've noticed if you select Web Socket Sharp instead of Web Socket NET in the Unity URDF Transfer advanced settings then more steps go from open to done.
[image]

In addition using Web Socket Sharp you see this in the terminal of the ROS vm -

> 2020-04-24 11:48:25+0100 [-] [INFO] [1587725305.504800]: Client connected.  11 clients total.
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.542962]: [Client 14] [id: /file_server/get_file:0] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.545175]: [Client 14] [id: /file_server/get_file:1] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.548476]: [Client 14] [id: /file_server/get_file:2] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.549416]: [Client 14] [id: /file_server/get_file:3] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.550386]: [Client 14] [id: /file_server/get_file:4] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.555578]: [Client 14] [id: /file_server/get_file:5] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.559114]: [Client 14] [id: /file_server/get_file:6] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.562349]: [Client 14] [id: /file_server/get_file:7] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [ERROR] [1587725305.565256]: [Client 14] [id: /file_server/get_file:8] call_service InvalidModuleException: Unable to import file_server.srv from package file_server. Caused by: No module named file_server.srv
> 2020-04-24 11:48:25+0100 [-] [INFO] [1587725305.919796]: Client disconnected. 10 clients total.
> 
> 

Expected results:
URDF is imported into Unity

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---

ChrisMOxon, 2020-04-24:
ROS version is Kinetic - not sure why that isn't showing

berkayalpcakal, 2020-04-24:
Hello @ChrisMOxon,

There is a known issue when using `WebSocketNet` protocol with `Microsoft` serializer as mentioned in #301, so I suggest that you keep using `WebSocketSharp` with `Microsoft` serializer.

Regarding the error `No module named file_server.srv`, are you using the latest version of ROS#? I think there is not a service definition called 'file_server.srv' in ROS# at the moment. I suggest that you pull the latest version of ROS#, place the package [file_server](https://github.com/siemens/ros-sharp/tree/master/ROS) under the src folder of your catkin workspace, build your catkin workspace, then source /catkin_ws/devel/setup.bash on your terminal, then try again.

I hope this helps.

ChrisMOxon, 2020-04-24:
Hi,

I'm on the latest version, I don't think I'm the only person who's had this problem though -

https://github.com/siemens/ros-sharp/issues/209

berkayalpcakal, 2020-04-24:
Sorry, I still think that there is something wrong with your catkin workspace environment. You and the other whom you referred probably did not managed to introduce the file_server package to your catkin workspace properly.

[This](https://answers.ros.org/question/52790/no-module-named-srv/) and [this](https://github.com/socrob/autonomous_systems/issues/4) perhaps can help.

I cannot reproduce the issue. If you can resolve the issue and think that this is a common issue with ROS#, you can comment your experiences here so that the community can benefit.

kkkkkaiai, 2020-04-27:
Hello, I have meet the issue before and now I solve it successfully. Maybe my experience can help you.

When I launch the `publish_description_turtlebot2.launch`, there are some warnings that means the URDF file can't be parsed correctly. So I change the file in line 24 to `<arg name="urdf_file" default="$(find xacro)/xacro --inorder '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'" />`, and the warnings disappear.

**[before]**
[image]

**[after]**
[image]

blo85, 2020-04-28:
I have exactly the same problem as @ChrisMOxon . And @kkkkkaiai solution doesn't work for me (I think these are different issues). Any ideas? @berkayalpcakal

Chrismmint, 2020-04-28:
Hi, so in the end I deleted all the folders I'd moved into my workspace, built workspace without them, reimported them from the copy I'd cloned, built workspace with them, re-sourced setup.bash, still wasn't working.

Turned VM on and off, started working.

blo85, 2020-04-29:
Thanks @Chrismmint ! Restart the VM worked for me. :D

MartinBischoff, 2020-04-29:
I consider this issue done.

closed 2020-04-29 by MartinBischoff