siemens/ros-sharp #310: Conflict with Unity's Nuget.Newtonsoft.Json implementation

coldwarrior5, 2020-05-12:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions, I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.
---
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all the required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 10 Pro Version: 1903 OS build: 18362.592
* My Unity Version is: Unity 2019.3.13
* My ROS Distribution is: ROS Melodic Morenia
* My Build Target Platform is: Windows

**Here is my bug description:**
The Unity DOTS packages now include Unity's implementation of Newtonsoft.Json and because of that I get errors in Unity, this is a recent development that occurred with new Unity package updates.
If I have the Ros# Newtonsoft.Json.dll in the Plugins folder I get an error that there cannot be two identical assemblies, so it is clashing with the Unity's one, but if I delete Newtonsoft.Json.dll assembly then I get an error that the Newtonsoft.Json.Bson.dll cannot find Newtonsoft.Json.dll, so it doesn't recognize Unity's implementation of Newtonsoft.Json.dll.

Perform the following steps to reproduce the bug:
Create a new empty Unity project, paste the latest Ros# code from GitHub into the Assets folder. Everything should work fine, then open Package Manager and install preview package called Collections (version preview.5-0.8.0). There will be an error saying there cannot be two identical assemblies in the Unity, if you delete the assembly then the error says that the Newtonsoft.Json.Bson.dll cannot load of find file or assembly Newtonsoft.Json which it should since Unity has its own implementation included under assemblies.

Observed results:
After installing Collections package:
[image]

After removing the Newtonsoft.Json.dll:
[image]

Expected results:
The Newtonsoft.Bson.Json.dll should be able to find the Newtonsoft.Json provided by Unity instead.

MartinBischoff, 2020-05-12:
Hi @coldwarrior5 ,
please re-complie the VS solution [RosSharp.sln](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosSharp.sln) with the dlls that work with your Unity project.
Alternatively (or for fast testing) you can copy the sources of the projects RosBridgeClient, MessageGeneration and Urdf directly into your Assets folder.
Please also check out the [ISerializer](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/Serializers/ISerializer.cs) interface. We designed it such that you can easily plug-in the Serializer of your choice.
Please let me know what worked best for you,
Martin

coldwarrior5, 2020-05-13:
Hi @MartinBischoff,
thank you very much for your fast reply, I have been trying all of your solutions from yesterday but I wasn't successful.
- If I copy the sources RosBridgeClient, MessageGeneration and Urdf to the Asset folder there is still issue with Newtonsoft.Json.dll and System.Runtime.CompilerServices.Unsafe.dll because Unity uses different versions and it disallows multiple instances of the same .dll. There are no problems with your RosBridgeClient.dll, MessageGeneration.dll, and Urdf.dll that you provide in Unity folder of your project. 
- If I try to overwrite the .dll in Unity's /Library/PackageCache/com.unity.collections@<version> with the ones that you provide the Unity overwrites it back to the original version.
- If I try downgrading the System.Runtime.CompilerServices.Unsafe in your source by using Nuget the code no longer compiles since it requires some functionality from newer version of Unsafe.dll.
- I tried importing Unity's Newtonsoft.Json.dll but it simply doesn't work with the other libraries, they might have changed something to work with Mono.
- ISerializer interface is extremely useful however it does not help with the library requirements, even if I try to remove Newtonsoft.Bson.Json and Newtonsoft.Json the code breaks because some classes depend on it and Unity does not allow two instances of the same libraries.

I discovered that other developers had the same issue I am having however, there doesn't seem to be any clean and nice solution currently [(Unity forum thread)](https://forum.unity.com/threads/conflict-between-assets-newtonsoft-and-packagemanagers-newtonsoft.843898/).
Thank you for your help, hopefully this will help you with your next version of Unity asset. When might we expect the next release in the Unity's Asset Store.

coldwarrior5, 2020-05-14:
Thanks to Unity's community I was able to find a solution, but I think it is the Unity's bug.

The solution to this particular problem is to manually add the `"com.unity.nuget.newtonsoft-json": "2.0.0"` to the manifest.json file since Unity doesn't do it automatically. The manifest.json file is located under the Project folder\Packages. Then you can safely delete the Newtonsoft.Json.dll from the Assets\RosSharp\Plugins\External folder and use the Unity's implementation. The Newtonsoft.Bson.Json.dll no longer complains about lack of Newtonsoft.Json.dll.

I think this is Unity's problem because for every other package that is shown in the Package manager there is an entry in the manifest.json, but for these "hidden" packages used in some of the global ones there is no entry in the manifest.json and because of that other .dll will no see it and thus there was an error in the Newtonsoft.Bson.Json.dll.

However, since Unity didn't implement specific package for System.Runtime.CompilerServices.Unsafe.dll the same cannot be done for that specific .dll but I will notify them about the issue.

coldwarrior5, 2020-05-15:
I found the solution for the second issue with the System.Runtime.CompilerServices.Unsafe.dll.

The sort of solution which is by no means ideal is to move the `com.unity.collections@<version>` folder from the Project folder\Library\PackageCache to Project folder\Packages, which makes that particular package a development package. Once copied over, we can now replace the System.Runtime.CompilerServices.Unsafe.dll in that folder with the newer version required by the Ros#.
Hope this will help you with the Unity development, I will notify Unity that they should expose the dll as a package the same way they did for the Newtonsoft.json.dll.

closed 2021-08-10 by MartinBischoff