siemens/ros-sharp #311: When I use Microsoft Serializer, will implicit conversion float to int.

Andor233, 2020-05-15:
When I use Newtonsoft Serializer, `9999.0` will be `9999.0` in the JSON.
But if use Microsoft Serializer, `9999.0` will be `9999`.
It causes many problems to me

Andor233, 2020-05-15:
[image]
[image]

Andor233, 2020-05-15:
integer and number is different in JSON

berkayalpcakal, 2020-05-15:
I could reproduce the issue. 

Using Newtonsoft Json serializer to serialize std_msgs/Float32 during a publication leads;
```"{\"topic\":\"topic_float32\",\"msg\":{\"data\":4.0},\"op\":\"publish\",\"id\":\"topic_float32\"}" ```

Using Microsoft Json serializer leads;
```"{\"topic\":\"topic_float32\",\"msg\":{\"data\":4},\"op\":\"publish\",\"id\":\"topic_float32\"}" ```

Referring to discussions [here](https://stackoverflow.com/questions/61316308/system-text-json-serializes-a-value-1-0-of-type-double-to-an-value-1-of-type-int) and [here](https://github.com/dotnet/runtime/issues/35195), it is a matter of serialization format followed by the developers. If you really want to use Microsoft Json serializer, you should think about using a custom converter. 

Just for curiosity, how does this create many problems for your project?

Andor233, 2020-05-15:
I don't know the details, but if I use Microsoft Json serializer, my program can't work.
So I check JSON of two  serialize, and the difference is this

Andor233, 2020-05-15:
And I am trying using make custom converter, but not succeed

Andor233, 2020-05-18:
Sorry I found it is my problem

closed 2020-05-18 by Andor233

MartinBischoff, 2020-05-18:
Hi @Andor233 thanks for the update. What was the problem? Is there anything others can learn from your findings?

Andor233, 2020-05-18:
My problem is just forgetting to regenerate the message after changing to the Microsoft JSON serializer.
And "3" is valid JSON for a number. But it depends your JSON implement, some works with "3", and some don't work. But ros bridge is working with both "3" and "3.0", it will convert implicitly by the ros message define.