siemens/ros-sharp #312: URDF transfer issues

uahic, 2020-05-15:
Hi,

I wrote a custom launch-script (modified version from the turtlebot  example of the file_server) which succesfully launches the web-bridge, started the URDF-ROS->Unity Importer connected to the localhost 127.0.0.1:9090 but after it connects to the webbridge (the bridge confirms '1 client connected'), it seems to stuck and unable to receive the URDF before it timesout. The launch script sucessfully has launched the file_server and runs xacro to read-in the specified URDF.

Versions:
Unity 2019.3.13f1
OS: Ubuntu 18.04
ROS-melodic
ROS#: master branch (head)

The log of Unity shows no exceptions or warnings beside from the debug-warning of the function ImportAssets:  "Not all resource files have been received before timeout".

What could have gone possibly wrong? Any suggestions how I might debug this further? 

Thank you very much

berkayalpcakal, 2020-05-15:
Hello @uahic ,
Please refer to the issues #301, #309.

MartinBischoff, 2020-05-22:
closing, see comment from @hahahuhu663 in issue #314.

closed 2020-05-22 by MartinBischoff