siemens/ros-sharp #313: Camera problems when importing Turtlebot URDF

sterlingm, 2020-05-16:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

When I run the GazeboSimulationScene in Unity, the Turtlebot 2 camera falls off (Roomba base) or moved from its intended pose (Kobuki base). 

I imported the URDF files, and then play the scene.

When using the Rooma base, the camera completely falls off the robot. Here is a picture: https://imgur.com/a/aFgO9S3

With the Kobuki base, the camera has not fallen off, but it does not have the correct pose: https://imgur.com/a/R5iN6Cg

I tried making the changes to ColladaAssetPostProcessor as advised in a similar issue here: https://github.com/siemens/ros-sharp/issues/47   but that did not change the camera pose.

I am using a Ubuntu 16.04 VM and Unity 2019.3.0f6. I imported ROS# from the Asset store.

MartinBischoff, 2020-05-18:
Hi @sterlingm ,
* could you please upload the URDFs here, to make sure we are talking about the same versions?
* Could you please also run Gazebo simulations with these URDFs and confirm that the behaviour is correct there? Is the camera position for Kobuki base correct in rviz?
* Please also feel free to check the URDFs yourself in the meantime:
  * in the Rooma base version, is there a fixed joint missing for the camera?
  * in the Kobuki base version, is the camera orientation defined at some "unusual" position in the URDF?

sterlingm, 2020-05-18:
Hi @MartinBischoff ,

I uploaded a zip of the URDF files. The files were created when installing the turtlebot packages on a clean install of Ubuntu 16.04 and ROS Kinetic. They have not been modified so I don't think anything is out of place or missing. I ran the turtlebot_world.launch file in turtlebot_gazebo for both robot models, and both of them were fine in Gazebo. I checked rviz, and the camera pose and camera_link frame both look good in rviz.

[urdfs.zip](https://github.com/siemens/ros-sharp/files/4645996/urdfs.zip)

berkayalpcakal, 2020-05-20:
Hello @sterlingm,

One check that came to my mind is that maybe there is a physics simulation running on Unity, which makes the rigidbodies fall down. If you are running the physics simulation in Gazebo, and subscribing to the joint states, odom, sensor data etc. from Unity, you should make sure that all rigidbody components of the robot in Unity are set to kinematic. You can ensure this by clicking on "Enable" button for the "IsKinematic" option via the inspector editor window when you click on the robot gameobject. And also please make sure that all the components are linked for the subscribers/publisher scripts under the "RosConnector" gameobject.

sterlingm, 2020-05-21:
@berkayalpcakal Yep, that's all it was! I just needed to enable the IsKinematic option for the robot's game object in Unity. Thanks for the help!

closed 2020-05-21 by sterlingm