siemens/ros-sharp #314: Cannot import URDF from ROS to Unity

hahahuhu663, 2020-05-19:
Hi.

* My OS is: Ubuntu 16.04
* My Unity Version is: 2019.3.13f1
* My ROS Distribution is: ROS-kinetic
* My Build Target Platform is : Ubuntu 16.04

I found a similar bug to issues #301 and #312. However, I tried to follow the instruction given in issue 301 but the bug remains the same.
I tried to import hector_quadrotor urdf to unity through rosbridge 3 months ago and it worked fine. However, I came back to tried again the issue occur after I update Unity3D's version from 2018.3.0f2 to 2019.3.13f1, so I have to create new Unity project and importing lastest version of RosSharp and file_server folder. I decided to launch the given launch file which is turtlebot2. 

The result is the software did connect with each other but it stuck at robot name.
[image]
[image]

Actually, I am not sure if I make a mistake anywhere. 

Thank you in advance.

berkayalpcakal, 2020-05-19:
Hello @hahahuhu663 ,
Please refer to the issues #301, #309.

hahahuhu663, 2020-05-19:
> Hello @hahahuhu663 ,
> Please refer to the issues #301, #309.

I read both of the issues, but 301's instructions didn't work. For 309's, I do not understand what to do with the code snippet you provided. It would be very kind of you if you could explain a bit more. I am quite new to this field.
Thank you in advance.

hahahuhu663, 2020-05-19:
> I read both of the issues, but 301's instructions didn't work. For 309's, I do not understand what to do with the code snippet you provided. It would be very kind of you if you could explain a bit [...]

Sorry a very late notice of my own unclear explanation. I try choosing websocket sharp from the drop down list, but it cannot be selected.

berkayalpcakal, 2020-05-19:
> Sorry a very late notice of my own unclear explanation. I try choosing websocket sharp from the drop down list, but it cannot be selected.

Interesting, have your tried changing it manually via the script [TransferFromRosEditorWindow.cs](https://github.com/siemens/ros-sharp/blob/2b217aafad238ae0e460bd2e20d9b1c99f76826c/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/Editor/UrdfTransfer/TransferFromRosEditorWindow.cs#L100)?

hahahuhu663, 2020-05-20:
> Interesting, have your tried changing it manually via the script [...]

I do not know what value should I put for protocol type in the script.

hahahuhu663, 2020-05-20:
I think it is all my faulty installation. I download ros sharp from asset store and everything work fine. Thank you for your help and your great work :).

closed 2020-05-20 by hahahuhu663