siemens/ros-sharp #315: Using this library with ROS2

maxlein, 2020-05-19:
I am not sure if this is a bug or just not supported.  
But as the timestamp msg has changed from "sec" to "secs" I can only use the ROS1 bridge and not the ROS2 version -> [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge/issues/145)  

Is this something known, or in the pipeline?

berkayalpcakal, 2020-05-19:
Hi @maxlein,

As far as I know, ROS# is not tested with ros2-web-bridge. If only problem is the naming of the message fields, then you can change the field names in the message classes manually on your local fork. For the timestamps in the header, [Timer.cs](https://github.com/siemens/ros-sharp/blob/2b217aafad238ae0e460bd2e20d9b1c99f76826c/Libraries/RosBridgeClient/BaseMessages/Std/msg/Time.cs) is the corresponding script where you can see the "sec" field. You just need to make the necessary changes, then rebuild the library and replace the dll's on the plugins folder. 

I hope that the difference in the naming convention is the only problem. Could you please right down your experiences regarding the use of ROS# with ros2-web-bridge? There is a plan to implement ROS2 support for ROS# soon, your experience could help us.

maxlein, 2020-05-20:
I only looked at some messages till now and they work with these small fixes:  

```diff
diff --git a/Libraries/RosBridgeClient/BaseMessages/Std/msg/Duration.cs b/Libraries/RosBridgeClient/BaseMessages/Std/msg/Duration.cs
index e7851bb..0db29d5 100644
--- a/Libraries/RosBridgeClient/BaseMessages/Std/msg/Duration.cs
+++ b/Libraries/RosBridgeClient/BaseMessages/Std/msg/Duration.cs
@@ -19,19 +19,19 @@ namespace RosSharp.RosBridgeClient.MessageTypes.Std
 	{
         public const string RosMessageName = "std_msgs/Duration";
 
-        public uint secs { get; set; }
-        public uint nsecs { get; set; }
+        public uint sec { get; set; }
+        public uint nanosec { get; set; }
 
         public Duration()
         {
-            secs = 0;
-            nsecs = 0;
+            sec = 0;
+            nanosec = 0;
         }
 
-        public Duration(uint secs, uint nsecs)
+        public Duration(uint sec, uint nanosec)
         {
-        	this.secs = secs;
-        	this.nsecs = nsecs;
+        	this.sec = sec;
+        	this.nanosec = nanosec;
         }
 	}
 }
\ No newline at end of file
diff --git a/Libraries/RosBridgeClient/BaseMessages/Std/msg/Time.cs b/Libraries/RosBridgeClient/BaseMessages/Std/msg/Time.cs
index 5ae121c..525a476 100644
--- a/Libraries/RosBridgeClient/BaseMessages/Std/msg/Time.cs
+++ b/Libraries/RosBridgeClient/BaseMessages/Std/msg/Time.cs
@@ -18,19 +18,19 @@ namespace RosSharp.RosBridgeClient.MessageTypes.Std
     public class Time : Message
     {
         public const string RosMessageName = "std_msgs/Time";
-        public uint secs { get; set; }
-        public uint nsecs { get; set; }
+        public uint sec { get; set; }
+        public uint nanosec { get; set; }
 
         public Time()
         {
-            secs = 0;
-            nsecs = 0;
+            sec = 0;
+            nanosec = 0;
         }
 
-        public Time(uint secs, uint nsecs)
+        public Time(uint sec, uint nanosec)
         {
-            this.secs = secs;
-            this.nsecs = nsecs;
+            this.sec = sec;
+            this.nanosec = nanosec;
         }
 
     }
```

alexandar1000, 2020-07-07:
Hello! 

My team and I are quite new to ROS and are just starting a project to connect it to Unity which is to be used as a simulator. Initially, our choice was to use ROS2 and make use of the fact that it provides some new functionality and will possibly be a bit more long-term solution. However, upon realising that ROS#, an important tool for connecting the two, does not fully support ROS 2, I am left puzzled on which version of ROS to use. 

Knowing this, and taking our unfamiliarity with the tools into consideration, would you say that it is better to use ROS1 until ROS# fully supports ROS2 and is tested sufficiently, or to use ROS2 anyways and adapt it as @maxlein suggested?  

Additionally, @maxlein, did you encounter any additional issues upon implementing this fix, or was that the only problem? 

Any advice would be immensely appreciated! Thank you in advance!

maxlein, 2020-07-07:
No, I had no other issues. Works fine and the fix is easy to implement.  
I may write a tutorial on ROS2 and Unity in the coming days.

alexandar1000, 2020-07-09:
Thank you for the update, really appreciate it!

DronistB, 2020-10-26:
> No, I had no other issues. Works fine and the fix is easy to implement.
> I may write a tutorial on ROS2 and Unity in the coming days.

Did you end up writing a guide on this subject? where can I find it?

sttobia, 2022-10-20:
> Hi @maxlein,
>
> As far as I know, ROS# is not tested with ros2-web-bridge. If only problem is the naming of the message fields, then you can change the field names in the message classes manually on [...]

Hello everyone,
Thanks for this library and the effort!
I just wanted to check in on the plans of implementing ROS2 support. Is this still planned in the near future, so that thechanges stated here are not required to be made whenever we use the library together with ROS2.
Thank you for an update on this topic.

Skoven, 2023-03-21:
Hi,

I have now compiled a version of the Visual Studio project that makes it ROS2 compatible. So far I have only looked at the communication itself, so changes to the RosBridgeClient.dll. The MessageGenerator.dll and the URDF.dll might already work for ROS2, but i have not checked that yet. 

I have created a fork of this repository with a branch called ros2: https://github.com/Skoven/ros-sharp/tree/ros2
Here is both the edited C# code, but also updated RosSharp folder for placing in the Unity project Asset folder:
com.siemens.ros-sharp/Assets/RosSharp

The issues fixed where:
1.  Fixing the timing labels. So going form `secs` to `sec` and `nsecs` to `nanosec`
2. Removed `.seq` from header as this is not a part of ROS2
3. Found that ROS2 puts "null" in unused parts of JointState msgs. The Json deserialiser of the ros-sharp library cannot handle this. I have just solved this by searching through the string an replacing "null" with 0.0

I hope that this will be of some help to poor souls like me, that struggle with getting ros-sharp to work with ROS2.

Dabuss, 2024-10-28:
Hi,

I recently discovered this library and really like it so far, the documentation is great, and I've found it very easy to use so far!

However, I just noticed that actions are not supported for ros2. Are there any plans to make it up to date with the advances in rosbridge_suite, for action support in ros2 (see https://github.com/RobotWebTools/rosbridge_suite/pull/886)?

Thanks!
David

memrecakal, 2024-10-30:
Hi @Dabuss,

Thank you for your interest. It's good to hear that you like it and find it easy to use. Yes, we are planning to add support for ROS2 actions. We didn't include it in the update because the way actions are handled has changed considerably from ROS1 to ROS2 and we want to take our time to implement it as properly as possible. I can't give you an estimated date for action support at the moment, but it's in our todo list.

Dabuss, 2024-10-31:
Hi @memrecakal, 

I've started working on it yesterday, and made good advances on a forked repo (action goal, feedback and result on the client side). I'm willing to contribute but I'd be happy to discuss this with someone, because I don't know how this can be done with my work constraints.
I just sent an email to ros-sharp.ct@siemens.com to discuss about it. Are you also in the loop?
David

memrecakal, 2024-10-31:
#460

closed 2024-10-31 by memrecakal