siemens/ros-sharp #316: Generating point cloud from depth & image values

Bradyk27, 2020-05-19:
Hi @NishanthJKumar ! Thanks for sharing this info about your project here!
I am looking forward to see the videos and images!

We did not publish our code for the turtlebot application open source. It would be too much effort to to make the code readable, document, fix bugs and administrated it properly. I hope you understand.

For the 3D point cloud we proceeded as follows:
* We used opencv library to uncompress the CompressedImage data.
* We used a script like [this](https://github.com/siemens/ros-sharp/blob/df13069a855329c0cfcf0a67ad02b2e214bfb983/RosBridgeClient/PointCloud.cs) to get the points in 3D from the two 2D images.
* We then put this data into a [mesh](https://docs.unity3d.com/ScriptReference/Mesh.html) with a simple shader that illustrates point data only.
* By only updating parts of the mesh (e.g. 20%) in each frame you can improve the performance of your application.
_Originally posted by @MartinBischoff in https://github.com/siemens/ros-sharp/issues/20#issuecomment-398302224_

Bradyk27, 2020-05-19:
Dr. Bischoff, could you expand upon this? I'm currently trying to do the same thing without much luck. 

I elected not to use opencv library to uncompress the data and instead just subscribed directly to the raw_image. This very well might be the source of my problem: did you run opencv on the Ubuntu side and then transfer the uncompressed images over? Or did you somehow run opencv within Unity?

Also, could you clarify this line?

`float depth = 0;// depthImage[u, v];`

What would depthImage[u,v] be? depthImage.data[u*v]?

Finally, where would one find / calculate the focal variable?

Thank you so much for your help--pulling my hair out trying to get this to work.

viliwonka, 2020-07-13:
I managed to sucessfully send point cloud through RosSharp, but it was very slow, it took like 20 seconds. But it was HD pointcloud, mind that, and I had to use BSON serialization since JSON itself doesn't support NaN values.

sterlingm, 2020-08-16:
@Bradyk27 I think using a DLLImport is the best (and only?) option to call OpenCV from Unity. I was able to do this to decompress the RGB images from an Asus Xtion Pro on a Turtlebot 2. Here is the C++ code I used:

`
extern "C"
{
	__declspec(dllexport) void processImage(unsigned char data[], int flag, int width, int height, int len, unsigned char** image)
	{
                // Create Mat to hold compressed data
		cv::Mat frame(height, width, CV_8UC3, data);

               // Decompress the data. Eventually use the "flag" parameter from processImage here.
		cv::Mat decompressed = cv::imdecode(frame, cv::IMREAD_COLOR);

		// image_transport sets encoding to bgr when the image has 3 channels, 
		// then converts from bgr to rgb with cvtColor
		// I was getting a grayscale image when not doing this. Not sure why 
		// Another option is set the Image encoding field to bgr8 (the image in rviz looked good), 
		// but it's probably better to keep the encoding consistent with what other ROS packages expect.
		cv::cvtColor(decompressed, decompressed, CV_BGR2RGB);

		// Set image data
		// len should be calculated and set in Unity
		*image = (unsigned char*)malloc(len);

		// Copy over the data
		memcpy(image, decompressed.data, len);
	}
}`

And in Unity I created this (unfinished) class to merge the RGB and depth images:

`
using UnityEngine;
using RosSharp.RosBridgeClient;
using System.Runtime.InteropServices;

public class RGDBMerger : MonoBehaviour
{

    [DllImport("image_decompress_opencv.dll", EntryPoint = "processImage")]
    public extern static void processImage(byte[] raw, int flag, int width, int height, int len, System.IntPtr image);

    public ImageSubscriber rgbImageSub;
    public ImageSubscriber depthImageSub;

    private RosSharp.RosBridgeClient.Messages.Sensor.Image rgbImage;
    private RosSharp.RosBridgeClient.Messages.Sensor.Image depthImage;

    // Add publishing info to check the decompressed images in Rviz
    public string Topic;
    private string publicationId;

    void Start()
    {
        publicationId = GetComponent<RosConnector>().RosSocket.Advertise<RosSharp.RosBridgeClient.Messages.Sensor.Image>(Topic);
        rgbImage = new RosSharp.RosBridgeClient.Messages.Sensor.Image();
        depthImage = new RosSharp.RosBridgeClient.Messages.Sensor.Image();
    }

    // Update is called once per frame
    void Update()
    {
        DecompressRGB();
        // DecompressDepth();....
    }

    protected void Publish(RosSharp.RosBridgeClient.Messages.Sensor.Image message)
    {
        GetComponent<RosConnector>().RosSocket.Publish(publicationId, message);
    }

    void DecompressRGB()
    {
        if (rgbImageSub.ImageData != null)
        {
            if (rgbImageSub.ImageData.Length > 0)
            {
                // Calculate number of elements in byte array. 3 channels in rgb images
                int len = (rgbImageSub.width * 3) * rgbImageSub.height;

                // Allocate managed memory array
                rgbImage.data = new byte[len];

                // Allocate unmanaged memory
                System.IntPtr mem = Marshal.AllocHGlobal(len);

                // Call dllimport function to fill in unmanaged memory
                processImage(rgbImageSub.ImageData, 1, rgbImageSub.width, rgbImageSub.height, len, mem);

                // Copy unmanaged memory into managed byte array
                Marshal.Copy(mem, rgbImage.data, 0, len);

                // Deallocate unmanaged memory
                Marshal.FreeHGlobal(mem);

                // Set image fields
                rgbImage.encoding = "rgb8";
                rgbImage.height = 480;
                rgbImage.width = 640;
                rgbImage.is_bigendian = 0;
                rgbImage.step = 1920;

                rgbImage.header.frame_id = "camera_rgb_optical_frame";

                // Test by publishing image and checking rviz. 
                // This is slow so only use it to test.
                //Publish(rgbImage);
           }
       }
   }
}
`

The `DecompressRGB` method seems to work to decompress the RGB images from an Asus Xtion Pro. I have not been able to figure out how to decompress the depth images yet though. Based on the code in the `image_transport` package, there's more involved in decompressing depth images than just calling `imdecode`, and I don't fully understand it all yet. The best solution, though, is to support `image_transport` in ROS#. A search for image_transport in the ROS# issues doesn't come up with much so I'm not sure if it's been attempted or not yet.

sterlingm, 2020-09-16:
@Bradyk27 Here is what worked for me to use OpenCV to decompress images coming from an Asus Xtion Pro on a Turtlebot2.

C++ side: https://pastebin.com/Ah7qL5yE
I could not get the `image_transport` codecs to compile in VS17 so I copied in code from that package to do the quantization for depth images. If your images are not encoded by `image_transport` then you may need to change some things. I wasn't sure if it was appropriate to add this code to the repo so for now I left it out. Also, keep in mind the `Mat` type and `imread` mode may be different depending on your image type. For my case, I was using jpg for RGB and png for depth.

C# side: I created a script to merge the 2D images, but right now it only decompresses. This script uses the DLL Import: [RGBDMerger.cs](https://github.com/sterlingm/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RGBDMerger.cs)
I had to make some minor changes to ImageSubscriber, and create a new script DepthImageSubscriber. The DepthImageSubscriber script is just a copy of ImageSubscriber, but it's necessary in order to have one component that can subscribe to two images. [ImageSubscriber](https://github.com/sterlingm/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/ImageSubscriber.cs) [DepthImageSubscriber](https://github.com/sterlingm/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/DepthImageSubscriber.cs)

The decompression takes between ~10-30ms for each type of image. You'll need to manually set the width, height, and msg fields of your images based on your camera+robot. Hope this helps.

sterlingm, 2020-09-22:
@Bradyk27 To get the focal length of your camera, check for `sensor_msgs/CameraInfo` data to be published on ROS. When working with a Turtlebot 2, for instance, the camera information is published on topics `/camera/X/camera_info` where X is depth, rgb, etc.

closed 2020-10-12 by MartinBischoff