siemens/ros-sharp #320: ROS# does not work when building the project on unity 2019.3.9f1

Changliu52, 2020-06-20:
Hi @berkayalpcakal 

Im on windows 10 using unity 2019.3.9f1
I have done the project setting with net.4x and pulling the latest master.
I have selected the microsoft and websocketsharp, following https://github.com/siemens/ros-sharp/issues/296

I also did the temp config following https://github.com/siemens/ros-sharp/issues/310

So far, no the build for windows x86_64 doen't work publish or subscribe anything. unity project works ok.

Any suggestions would really be appreaciated.

berkayalpcakal, 2020-06-20:
Hello 

Could you please use the issue template, and provide more info about the build errors or so?

Changliu52, 2020-06-20:
Sure, just edited the issue description.
there is no build/run error. there are some warnings, but nothing particularly related to rossharp. There is one warning im not sure if it is related.
```
System.Windows.Forms.dll assembly is referenced by user code, but is not supported on StandaloneWindows64 platform. Various failures might follow.
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)
```
other than that it is a successfull build.

Changliu52, 2020-06-20:
I have installed the melodic-rosbridge-server_0.11.9-1bionic.20200530.142154_amd64 in the ROS side. The ros side detects the client connection. But output the below error:
```
publish: cannot infer topic type for topic /arm_controller as it is not yet advertised.
```
and no topic was published. This happends for the build app, but the unity project runs smoothly.

berkayalpcakal, 2020-06-21:
The problem of not publishing/subscribing to any topics even though rosbridge_server detects a connected client has been seen only with the use of WebSocketNet protocol so far, to  the best of my knowledge (see #309, #301).

Which protocol and serializer are you using? If you feel like the problem is somehow related to function calls from the RosBridgeClient.dll, you can remove it from Assets/RosSharp/Plugins folder and import the library scripts into Assets folder, and fully debug the ROS#.

Changliu52, 2020-06-22:
I was using websocketsharp protocol and microsoft serialiser.
I have solved the problem by reimporting the rossharp 1.6 release and selecting the wesocketsharp and json. Seems working now.

closed 2020-06-22 by Changliu52