siemens/ros-sharp #321: Planning in Unity using ROS Moveit

abinet0TS, 2020-06-25:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## Planning in Unity using ROS Moveit

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
Hello, 
I'm currently working on the Moveo BCN3D robot and I want to know if it possible to use ROS# to move the interactive marker from Moveit in Unity? On the same topic can i receive the visual of the Goal State instead of the robot visual. 

Thank you in advance for your awnser.
---

* My OS is: <Windows 10 for Unity and Ubuntu 16.04 on a VirtualBox>
* My Unity Version is: <2019.3.15>
* My ROS Distribution is: <Kinetic>

berkayalpcakal, 2020-06-25:
Hello @abinet0TS,

I think that the interactive marker is a part of MoveIt plugin for Rviz, and I am not sure whether you can actually move the marker without actually interacting with the plugin by using your mouse. 

If what you want to do is simply setting target goals for MoveIt from Unity, then you should find a way to communicate with the move group node which should have already been running when you launched the entire MoveIt routines on the ROS side. Please see MoveIt tutorials for that. The way of communicating from Untiy to the move group node totally depends on you. The way I did in [my repo](https://github.com/berkayalpcakal/ROS-HoloLens-Integration/blob/master/RosPackages/src/moveit_tutorials/doc/move_group_python_interface/scripts/mar_master.py) was to set up some action servers on the ROS side and have the action clients on the Unity side. You can check out the ROS# wiki pages on setting up action clients on the Unity side. 

I hope this helps.

closed 2020-06-25 by berkayalpcakal

abinet0TS, 2020-06-29:
Hello,

thank you for your quick answer. I 'm a beginner in ROS and have a few
difficulties. I don't want to take too much of your time, but if you could
give me a few more explanations on setting an action server that sets a
goal to reach in ROS i would really appreciate it.

Best regards,
BINET Antoine

MartinBischoff, 2020-06-29:
Dear @abinet0TS,
http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionServer%28ExecuteCallbackMethod%29
This is no ROS helpdesk. Please mind the checkboxes you marked up in the issue template.
Best,
Martin