siemens/ros-sharp #326: Linux Unity Editor support

jbga, 2020-07-05:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case. 

## Above box not checked because a blank page appeared when following the link 
---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Does this package supports Unity Editor running on Linux?
After importing this package and press play all the objects scripts references dissapeared and missing dlls errors appeared on the console.

berkayalpcakal, 2020-07-05:
Hello @jbga,

If your search the issue board thoroughly, you can see that some people actually got ROS# working within Unity on Linux. See for example #296, #297. 
Please give more detailed explanation regarding the issue you have.

jbga, 2020-07-06:
Erros like this appears when I import the project:
```
Assembly 'Library/ScriptAssemblies/Assembly-CSharp.dll' will not be loaded due to errors:
Reference has errors 'RosBridgeClient'.

Assembly 'Library/ScriptAssemblies/Assembly-CSharp-Editor.dll' will not be loaded due to errors:
Reference has errors 'Assembly-CSharp'.

Assembly 'Assets/RosSharp/Plugins/RosBridgeClient.dll' will not be loaded due to errors:
RosBridgeClient references strong named System.Threading.Tasks.Extensions, versions has to match. Assembly references: 4.2.0.1 Found in project: 4.2.0.0.

Assembly 'Assets/RosSharp/Plugins/Microsoft/System.Text.Encodings.Web.dll' will not be loaded due to errors:
Reference has errors 'System.Memory'.
```

* My OS is: Ubuntu 18.04.4 LTS
* My Unity Version is: 2020.1.0b9

berkayalpcakal, 2020-07-06:
Your issue looks very similar to #300, maybe the procedures there would help you.
Otherwise I suggest that you try ROS# with the latest LTS version of Unity.

Please carefully read what you confirm on which you put a tick, and indeed do search the closed issues!

> * I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer.

closed 2020-07-06 by berkayalpcakal