siemens/ros-sharp #327: IProtocol implementation that does not use WebSocket

OrangeJuice7, 2020-07-06:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a feature request! ##

**I want to do this:**

I currently need to implement ROS support (in particular, publishing to a topic so I can .bag dump the messages) for a game written in C# (NOT Unity). After searching the internet, my only choices to communicate to ROS from a C# program seem to be either ROS#, or [roscs](http://wiki.ros.org/roscs) (which has been deprecated after ROS Fuerte, among other issues).
I would like to use ROS# for my use case, but [ROS# has only implemented WebSocket protocols](https://github.com/siemens/ros-sharp/wiki/Dev_Protocols) - even though it is stated that it does not necessarily have to be so, which I find creates unnecessary dependencies (especially needing to connect to the "internet") for communication within the same machine.

**For being able to do it, I wish ROS# had the following feature:**
ROS# could have an implementation of `IProtocol` that is meant for internal communication within the same machine, and does not depend on `WebSockets` and *its* dependencies (like a URI).
Because I would like this functionality soon, I might implement this myself, if I may be pointed in a direction.

Disclaimer: I am new to ROS and its libraries like rosbridge, so please pardon me if I do not understand or am not aware of of certain things.

Thank you very much!

sagivald, 2020-08-20:
If you are working with ROS2 , what you are looking for is in development in partially working [Ros2DotNet](https://github.com/ros2-dotnet/ros2_dotnet) see example from [DynoRobotics-Unity2ROS](https://github.com/DynoRobotics/UnityRos2) , if you are working with ROS1 its not possible to implement an internal communication becuase its not supported. your options is building one from scratch or use RosBridgeSuite which is implemented over WebSockets and can be done by ROS# a state of the art tool. or a different one that use Protobuf such as [LGSVL-RosBridge](https://github.com/lgsvl/simulator). it would be nice if the fellas here of ROS# community will develop a Protobuf Serializer for ROS#

MartinBischoff, 2020-10-07:
I totally agree with @sagivald s comment above. In addition you could also think about using interprocess communication, i.e. [NamedPipes](https://docs.microsoft.com/en-us/windows/win32/ipc/named-pipes). The corresponding interface on the ROS side would have to be build from scratch as already mentioned above.

closed 2020-10-07 by MartinBischoff

OrangeJuice7, 2020-10-08:
Hello, thank you for the replies!
I was indeed using ROS1.

Just to share my final solution, I did not use ROS# at all, and simply made my application start up a custom ROS node as a child process, and communicated with the node from the parent app via the child's stdio.

moisc18, 2021-12-20:
Hi @OrangeJuice7 I'm tryin to do a project similar to yours, and I need to create a child ROS node too. Can you share more information about how did you do it?

markdav-is, 2022-01-06:
Hi @OrangeJuice7 and @moisc18 I'm in the same boat.  Looking to integrate ROS into .Net 6, (core) without unity or windows.  I'm running on linux.