siemens/ros-sharp #329: Add support for Audio messages

Chambana, 2020-07-22:
* [X ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.
## I have a feature request! ##

**I want to do this:**
Receive ROS audio data from my remote robot and play it back in my computer speakers.

**For being able to do it, I wish ROS# had the following feature:**
Please add support for sending/receiving ROS audio data, to include sample playback C# code.
https://github.com/ros-drivers/audio_common/blob/master/audio_common_msgs/msg/AudioData.msg

berkayalpcakal, 2020-07-22:
Hello @Chambana,

Looking at the AudioData.msg, it seems like data is just an array of bytes which can easily be transferred from ROS to Unity via custom subscriber script on Unity side. To do so, you need to have an equivalent message definition of AudioData.msg on Unity as c# script (please see this [wiki page](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes) ), and a subscriber script that inherits [UnitySubscriber.cs](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnitySubscriber.cs)  (please see the list of example subscriber scripts [here](https://github.com/siemens/ros-sharp/tree/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation)). 

Once you have the audio data transferred from ROS to Unity, it is a matter of choice how you interpret and process the data. I think that the Unity Forum is the best place to find answers at this point.

If you would have a nice way of doing it, you are welcome to share your experiences here.

MartinBischoff, 2020-07-23:
Hi @Chambana ,

In addition to @berkayalpcakal comments above, I have the following hints that I just found by searching a bit about streaming audio from/to Unity:
* For both playing and recording sounds in Unity the builtin class for managing audio data is [AudioClip](https://docs.unity3d.com/Manual/class-AudioClip.html).

* [Here](http://bubblestorm-dev.blogspot.com/2016/06/realtime-sound-generation-in-unity3d.html) is a good code example for streaming sounds to Unity with builtin components. In the PCMReaderCallback. In the example some audio data is generated in the `PCMReaderCallback`, whereas in our case the data is received and buffered by your `AudioDataSubscriber` implementation of [UnitySubscriber](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnitySubscriber.cs) and the callback continuously reads from that buffer.
* To publish data via ROS# it is best practise to implement a corresponding [UnityPublisher](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnityPublisher.cs) class, which I'd name `AudioDataPublisher` in you case and that continuously recevies the audio data from another another [PCMReaderCallback](https://docs.unity3d.com/ScriptReference/AudioClip.PCMReaderCallback.html) implementation.
* If you want to stream data via Microphone from Unity, [here](https://docs.unity3d.com/2017.3/Documentation/ScriptReference/Microphone.html) is the corresponding builtin Unity class that provides the microphone recordings to `AudioClip`.

* In my comments above I didn't consider the actual audio data formats supported by ROS and Unity. You might need additional converters, and possibly external dependencies. [Here](https://assetstore.unity.com/packages/tools/audio/audiostream-pcm-audio-in-unity-65411) is a Unity Asset dedicated to Audio Streaming with Unity.

If we find some simple and generic implementation for sending and receiving audio data from and to ROS that doesn't require additional dependencies, I think it would be some helpful addition to ROS#.

ookayt, 2023-08-12:
> * [X ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack [...]

Hey, have you had any success finding an implementation in Unity for sending and receiving audio to and from ROS?
I have a similar use case.

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff

PeterBowman, 2025-03-27:
For future reference, I managed to transmit audio to and from the ROS network in a Unity VR app context (Meta Quest 2 headset), optionally using MP3 compression (via [NAudio](https://github.com/naudio/NAudio) + [NLayer](https://github.com/naudio/NLayer)) when receiving data: [transmission](https://github.com/roboticslab-uc3m/tiago-unity-vr/blob/60f21f437081a4cb4a5c120d3f408b08e6b3074e/Assets/Scripts/AudioRecorder.cs), [reception](https://github.com/roboticslab-uc3m/tiago-unity-vr/blob/60f21f437081a4cb4a5c120d3f408b08e6b3074e/Assets/Scripts/AudioPlayer.cs). My code relies on the [ROS-TCP-Connector](https://github.com/roboticslab-uc3m/ROS-TCP-Connector) project (no longer maintained, but usable); however, the algorithm boils down to proper data handling, so it should be extensible to other frameworks.