siemens/ros-sharp #333: Why the  turtlebot2 is in a mess

329937626, 2020-08-11:
[image]
I have do the Tutorial:
2.1 Transfer a URDF from ROS to Unity
But,the robot is in a mess when the "Play" Button is click, like above screenshot.
Why??

329937626, 2020-08-11:
The Kinect is fall down on the ground , the turtlebot2 's parts is separate.
 The turtleot2 is scampering, like in a mess. Why?

329937626, 2020-08-11:
unity version：
2018.4.25  or 2019 is the same result why?

EricVoll, 2020-08-11:
Do you get any Errors in the Unity Console?
I had something similar once where the code crashed after creating the game objects but before setting "IsKinetic" to true.

329937626, 2020-08-12:
> 
> 
> Do you get any Errors in the Unity Console?
> I had something similar once where the code crashed after creating the game objects but before setting "IsKinetic" to true.

Thank you very much! Your method worked, thank you.

closed 2020-08-12 by 329937626

jiayiyi1991, 2020-08-12:
I use the last code , I have the same problem. Set "IsKinetic" to true, turtlebot can‘t move ; Set "IsKinetic" to false, turtlebot broke；When I change last releases version 1.6 ,  everything is ok；

329937626, 2020-08-12:
> I use the last code , I have the same problem. Set "IsKinetic" to true, turtlebot can‘t move ; Set "IsKinetic" to false, turtlebot broke；When I change last releases version 1.6 ,  everything is ok；

My version is last releases. pull it a few days ago.

329937626, 2020-08-12:
你用的是哪个版本的教程？你配置的Controller 配置是这样的吗

berkayalpcakal, 2020-08-12:
@329937626 
Why are you writing something in Chinese? 
Do you even read the comments above?

> Do you get any Errors in the Unity Console?
> I had something similar once where the code crashed after creating the game objects but before setting "IsKinetic" to true.

Just set the IsKinetic flag true on the inspector editor of your robot!