siemens/ros-sharp #337: Publishing a boolean variable from Unity to Ros topic

andreb-88, 2020-08-12:
## I have a question! ##
* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I am quite new in unity. I need to synchronize my recording video on Unity with the data frame analysis in ROS with RosConnector. I thought to create a boolean variable to publish in a ROS topic for notifying to my nodes that my recording in unity it stared. It may be shameful but I can't understand how to do it. I tried to create a C# script to handle the publishing but i don't know how to get my boolean variable. I'm sorry if i did't understand something it would be obvious in the wiki.
I attached here the script that i wrote:

namespace RosSharp.RosBridgeClient
{
    public class Status : UnityPublisher<MessageTypes.Std.Bool>
    {
        public bool messageData;

        private MessageTypes.Std.Bool message;

        protected override void Start()
        {
            base.Start();
            InitializeMessage();
        }

        private void InitializeMessage()
        {
            message = new MessageTypes.Std.Bool
            {
                data = messageData
            };
        }

        private void Update()
        {
            message.data = messageData;
            Publish(message);
        }
    }
}

MartinBischoff, 2020-08-14:
Hi @andreb-88 ,
without having tested it, your class looks correct. All you have to do now is to set your 'messageData ' boolean value.
Since 'messageData' is a public field you can set directly via the unity inspector, or from any other method in this class or some other.

closed 2020-08-14 by MartinBischoff

andreb-88, 2020-08-14:
Thank you! It works. I attached the script if it helps someone else.

using System.Collections;
using UnityEngine;

namespace RosSharp.RosBridgeClient
{
    public class DroneArmedPublisher : UnityPublisher<MessageTypes.Std.Bool>
    {
        private bool status;
        private MessageTypes.Std.Bool message;

        protected override void Start()
        {
            base.Start();
            InitializeMessage();
        }

        private void FixedUpdate()
        {
            status = UDPControl.GetDroneArmed();
            //Debug.Log(status);
            UpdateMessage();
        }

        private void InitializeMessage()
        {
            message = new MessageTypes.Std.Bool
            {
                data = status
            };
        }

        private void UpdateMessage()
        {
            message.data = status;
            Publish(message);
        }
    }
}